zephyrproject-rtos/zephyr #12440: Device discovery of direct advertising devices is not working

pdunaj, 2019-01-11:
**Is your feature request related to a problem? Please describe.**
Device discovery for direct advertising devices is not working.

**Describe the solution you'd like**
I would like device that are direct advertising to be discoverable.

**Describe alternatives you've considered**
I really need direct advertising so there are no alternatives.

**Additional context**
Today I tried to use direct advertising between peripheral and central running Zephyr. It seems nobody has used that for some time. There was a bug on peripheral side (https://github.com/zephyrproject-rtos/zephyr/pull/12438). After fixing the issue I see direct advertising packets going out of the device. However central does not see the peripheral (yes, device was bonded earlier).

I checked the code and I see that although normal advertising report is supported (look for `BT_HCI_EVT_LE_ADVERTISING_REPORT` in `hci_core.c`), direct advertising is not (lack of reaction to `BT_HCI_EVT_LE_DIRECT_ADV_REPORT`).

I have tried to put logs in `le_advertising_report` to see if any devices are actually found (even if report is not processed further) but I see no direct advertisement detected (I see packets on the sniffer that contain both my peripheral and central addresses). Has anybody actually used this?

pdunaj, 2019-01-11:
Hi @cvinayak , do you know if this scanning should work? It seems that there is code missing there...

carlescufi, 2020-03-17:
@cvinayak @pdunaj @joerchan is this still applicable?

joerchan, 2020-03-17:
I suspect this was the issue:

```
 *  Note: The LE scanner by default does not use the Identity Address of the
 *        local device when :option:`CONFIG_BT_PRIVACY` is disabled. This is to
 *        prevent the active scanner from disclosing the identity information
 *        when requesting additional information from advertisers.
 *        In order to enable directed advertiser reports then
 *        :option:`CONFIG_BT_SCAN_WITH_IDENTITY` must be enabled.
```
AFAIK adv reports with directed advertiser works. Given that one of the two options are enabled.
Using BT_PRIVACY is the one I would recommend in this case.

joerchan, 2020-03-19:
Closing this since I have confirmed with @MarekPieta That discovery of directed advertiser is working.

closed 2020-03-19 by joerchan

MarekPieta, 2020-03-19:
I verified that information about direct advertising packets is properly forwarded to the application through the registered callback. The same callback is used for indirect advertising packets (the advertising type field value is the difference). I tested with the nrfxlib LL.