zephyrproject-rtos/zephyr #12449: Existing LPN handling in Mesh Friend Request handling

isundaylee, 2019-01-12:
**Describe the bug**

Bluetooth Mesh Specification (3.6.6.3.1) says that 

> If the source address of the Friend Request message is an address of a Low Power node that is currently in a friendship with the Friend node, then the Friend node shall also consider the existing friendship with that Low Power node terminated.

This is currently handled here: https://github.com/zephyrproject-rtos/zephyr/blob/73956a398ed6cbd943a54bea7dc9c311ed0e2177/subsys/bluetooth/host/mesh/friend.c#L850-L863 

However, when the `prev_addr` field is set on the Friend Request message, the current code looks up existing LPN entry using the `prev_addr` field as the address. However, `prev_addr` contains the address of the requesting node's _previous friend_, **not** the address of the requesting node _itself_. This looks a bit weird - shouldn't we always look up existing entry using `rx->ctx.addr` as the address? 

In my case, this causes friendship retries to fail, since the previous lingering LPN entry is not properly cleared.

If I make this quick fix, https://github.com/isundaylee/zephyr/commit/e4ef4ea0eefa757f68ad3b53b0e7f0c844ca08e2, the issue goes away, and LPN retries work as expected. 

**To Reproduce**

N/A. This issue is a question about current code implementation.

**Expected behavior**

See above.

**Impact**

I can workaround using the aforementioned fix.

**Screenshots or console output**

N/A

**Environment (please complete the following information):**
 - OS: macOS
 - Toolchain: arm-none-eabi-gcc
 - Commit SHA or Version used: 73956a398e

**Additional context**

N/A

isundaylee, 2019-01-12:
I feel like #11682 could potentially be related.

jhedberg, 2019-01-12:
@isundaylee You're right that the logic doesn't make sense - it seems to be treating the prev_friend as if it was the address of an LPN. That said, there's a reason the field exists in the Friend Request. I'll need to double-check what we're supposed to do with it (in case this is the only place in the code that we're doing something with it)

jhedberg, 2019-01-12:
From section 3.6.6.3.1 Friend establishment:
> After a friendship has been established, if the PreviousAddress field of the Friend Request message contains a valid unicast address that is not the Friend node’s own unicast address, then the Friend node shall begin sending Friend Clear messages (see Section 3.6.5.5) to that unicast address according to the procedure below:
> 1. The TTL shall be set to the maximum valid value.
> 2. The first Friend Clear message shall be sent as soon as the friendship is established; at the same time, a Friend Clear Repeat timer shall be started with the period set to 1 second, and a Friend Clear Procedure timer shall be started with the period equal to two times the Friend Poll Timeout value.
> 3. If a Friend Clear Confirm message (see Section 3.6.5.6) is received in response to the Friend Clear message, both timers shall be canceled and the procedure is complete.
> 4. If the Friend Clear Repeat timer expires, a new Friend Clear message shall be sent and the timer shall be restarted with a period that is double the previous Friend Clear Repeat timer period. For example, after the first expiration, the period shall be set to two seconds; on the next expiration, it shall be set to four seconds, and so on.
> 5. If the Friend Clear Procedure timer expires, then the Friend Clear Repeat timer shall be cancelled and the procedure is complete.

I believe this is what the implementation is currently missing. Your fix that you referred to looks correct, so please send a pull request for that, however as a follow-up we should look at implementing the above procedure as well.

isundaylee, 2019-01-12:
@jhedberg Thanks for the quick response! I'll send in a PR for that commit. It looks like the Clear Procedure is already implemented, and is handled by https://github.com/zephyrproject-rtos/zephyr/blob/73956a398ed6cbd943a54bea7dc9c311ed0e2177/subsys/bluetooth/host/mesh/friend.c#L890-L893

isundaylee, 2019-01-12:
Submitted PR #12450.

jhedberg, 2019-01-12:
Thanks @isundaylee . Your PR is now merged, and you're right that the clear procedure is indeed in place - I just missed it since it's further down the same function.

jhedberg, 2019-01-12:
Closing as fixed by PR #12450

closed 2019-01-12 by jhedberg