zephyrproject-rtos/zephyr #12453: nrf52 SPIM spi_transceive function occasionally doesn't return

nwsetec, 2019-01-12:
**Describe the bug**
Occasionally a call to spi_transceive() doesn't return.  When this happens the system has to be reset to function again.  Sometimes when this occurs the Bluetooth submodule (which is also used in this project) and the shell continues to operate and other times they have both stopped working.

What have you tried to diagnose or workaround this issue?
I've looked at the each thread stack usage and there doesn't seem to be an issue there.  I've got the CONFIG_STACK_SENTINEL=y and CONFIG_STACK_CANARIES=y and neither of these detect failures while running.  I've tried increasing and decreasing the priority of the thread that contains the spi_transceive call to above and below the Bluetooth threads but that didn't help.

After searching through Zephyr issues I noticed issue #12342 and PR #12400 and wondered if this is related so I tried to cherry pick those commits to the 1.13.0 release, I got as far as the first 2 commits but the third (845e3835f06d892d4bd7ac3493c1ac4fc8e04d12) had too many changes since 1.13.0 to be easy to apply.  The two commits I could apply plus configuring CONFIG_SWAP_NONATOMIC=y didn't fix the issue.

**To Reproduce**

CONFIG_SPI=y
CONFIG_SPI_INIT_PRIORITY=70
CONFIG_SPI_0=y
CONFIG_SPI_0_NAME="SPI_0"
CONFIG_SPI_0_OP_MODES=1
CONFIG_SPI_0_IRQ_PRI=7
CONFIG_SPI_NRFX=y
CONFIG_SPI_0_NRF_SPIM=y
CONFIG_SPI_0_NRF_SCK_PIN=14
CONFIG_SPI_0_NRF_MOSI_PIN=16
CONFIG_SPI_0_NRF_MISO_PIN=11
CONFIG_SPI_0_NRF_ORC=0x00

`
static struct spi_buf 			stSpi_tx_buffs[3];
static struct spi_buf 			stSpi_rx_buffs[3];

static const struct spi_buf_set stSpi_tx_buff_array = {
	.buffers = stSpi_tx_buffs,
	.count = 3
};

static const struct spi_buf_set stSpi_rx_buff_array = {
	.buffers = stSpi_rx_buffs,
	.count = 3
};

void app_spi_init(void)
{

	stSpim_dma = device_get_binding(SPIM_DRV_NAME);

	stConfig_chipSelect.gpio_dev = stGpio_dev;
	stConfig_chipSelect.gpio_pin = SPI_CS_PIN;
	stConfig_chipSelect.delay = 0;

	stConfig_Spim.frequency = 1000000;
	stConfig_Spim.operation = (SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB
			| SPI_WORD_SET(8));
	stConfig_Spim.slave = 0;
	stConfig_Spim.cs = &stConfig_chipSelect;

	uint8_t *au8_tx_data;
	uint8_t *au8_rx_data;

	au8_tx_data = (uint8_t*)&SPIS_stuMData;
	au8_rx_data = (uint8_t*)&SPIS_stuSData;

	stSpi_tx_buffs[0].buf = &au8_tx_data[0];
	stSpi_tx_buffs[0].len = 200;
	stSpi_tx_buffs[1].buf = &au8_tx_data[200];
	stSpi_tx_buffs[1].len = 200;
	stSpi_tx_buffs[2].buf = &au8_tx_data[2 * 200];
	stSpi_tx_buffs[2].len = 200;

	stSpi_rx_buffs[0].buf = &au8_rx_data[0];
	stSpi_rx_buffs[0].len = 200;
	stSpi_rx_buffs[1].buf = &au8_rx_data[200];
	stSpi_rx_buffs[1].len = 200;
	stSpi_rx_buffs[2].buf = &au8_rx_data[2 * 200];
	stSpi_rx_buffs[2].len = 200;

	// Load SPI TX buffer into TWI module. Set number of bytes to write pr transfer
	NRF_SPIM0->TXD.PTR = (uint32_t)&au8_tx_data;
	NRF_SPIM0->TXD.MAXCNT = 200;
	NRF_SPIM0->TXD.LIST = SPIM_TXD_LIST_LIST_ArrayList
			<< SPIM_TXD_LIST_LIST_Pos;

	// Point to SPI RX buffer. Set number of bytes to read pr transfer
	NRF_SPIM0->RXD.PTR = (uint32_t)&au8_rx_data;
	NRF_SPIM0->RXD.MAXCNT = 200;
	NRF_SPIM0->RXD.LIST = SPIM_RXD_LIST_LIST_ArrayList
			<< SPIM_RXD_LIST_LIST_Pos;
}

`

The spi_transceive() call is happening at 20ms intervals.
`
spi_transceive(stSpim_dma, &stConfig_Spim, &stSpi_tx_buff_array, &stSpi_rx_buff_array);
`

**Expected behavior**
I've observed it on the CRO and can see that the SPIM operation completes (SS pin goes high, SPI clk signal stops) but a pin (used for debugging) that is set to high before the transfer and low after the transfer completes goes high and normally low afterwards but when the issue occurs it never goes low.

After up to an hour of operation the code stops working.

**Impact**
Showstopper

**Screenshots or console output**
[image]
In the above screenshot the blue falling pulse on the left shows a successful call to spi_transceive() and the one in the middle shows a successful transfer but no return to the calling thread.  The blue trace shows the SS line.  The green shows some post processing after the transfer is complete.  The yellow shows the debug pin described above going high just before the transfer and low afterwards.  Note the yellow trace also goes high and low again immediately after the transfer in the pulse on the left (redundantly showing the same information as the green trace).

**Environment (please complete the following information):**
 - OS: Windows 7
 - Toolchain: gcc-arm-none-eabi-7-2018-q2-update-win32-sha2.exe
 - Zephyr 1.13.0

**Additional context**
We're using the nrf52832 SoC.
Bluetooth is configured as a peripheral and observer.  It is configured to always be passively scanning.

carlescufi, 2019-01-13:
@nwsetec Thanks for the detailed report. It would really help if you could try to narrow down where inside `spi_transcieve` the code is blocked. You can use your blue debug pin for that, or another one.

DeclanTraill, 2019-01-14:
On further investigation, the call to spi_context_wait_for_completion() never returns.
Within this function it is the call to k_sem_take(&ctx->sync, K_FOREVER); that never returns.

If the sync semaphore give and take are replaced with a global volatile boolean flag, then there is no problem & the code runs (tested up to 2.5 hours so far with no problem).

e.g.

In spi_context_wait_for_completion(), instead of k_sem_take(&ctx->sync, K_FOREVER); do this:

while (!spim_done) k_sleep(1);

And in spi_context_complete(), instead of k_sem_give(&ctx->sync); do this:

spim_done = true; 

Where spim_done is defined as:

volatile bool spim_done = false;

Also:  shouldn't all semaphore variables that are shared between ISRs and Threads be declare as volatile, otherwise a thread may be waiting on a cached variable, rather than the actual value changed in an ISR?

Regards,
Declan Traill

DeclanTraill, 2019-01-14:
... Oh, and at the start of transfer_next_chunk(), add spim_done = false;

BTW: I tried just changing both the lock and sync semaphore variables in struct spi_context to volatile (which generated about 20 compiler warnings as Zephyr Macros are not expecting the volatile keyword, so drop it). This seemed to make the timing of the SPI processing more reliable (we were getting occasional CRC errors before) and I didn't see any CRC errors, but the code still stopped on the k_sem_take(&ctx->sync, K_FOREVER); 25 minutes later.

nwsetec, 2019-01-14:
@carlescufi if you hadn't already guessed Declan and I are working on the same issue :)

nwsetec, 2019-01-14:
@carlescufi I checked also and added pin toggles either side of spi_context_wait_for_completion() in the transceive() function and I see the toggle before but not after.

nwsetec, 2019-01-14:
For clarity I've removed the junk initialisation code (raw register access) that the nrf SPI driver would have overridden.

DeclanTraill, 2019-01-14:
My test using a volatile global flag instead of a semaphore has been running without problems for 20.5 hours, so it is a workaround for the time being, but not a proper solution as when I do a sleep(1) after checking the done flag, the check won't be performed again for ~6 msec. This could delay the post processing of the SPI data where a semaphore would cause the context switch to occur when the DMA has actually completed (or shortly thereafter).

nwsetec, 2019-01-15:
Which reminds me that our system clock tick is currently set as CONFIG_SYS_CLOCK_TICKS_PER_SEC=333 which explains the 6ms delay mentioned in Delcan's comment above (x2 3ms periods).
I changed CONFIG_SYS_CLOCK_TICKS_PER_SEC back to 50 (20ms period) and the issue still happened at the same spot.

DeclanTraill, 2019-01-17:
Further to earlier tests, I have determined that the k_sem_give(&ctx->sync); in spi_conext complete() DOES get executed prior to the failure, but the k_sem_take(&ctx->sync, K_FOREVER); never receives it.

Here is the console output form a test resulting in failure:

0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5 7
0 1 2 3 6 1 2 3 1 2 3 1 4 5

The numbers printed out are stages of the process:   Here is the code that generated them:

static int transceive(struct device *dev,
		      const struct spi_config *spi_cfg,
		      const struct spi_buf_set *tx_bufs,
		      const struct spi_buf_set *rx_bufs)
{
	struct spi_nrfx_data *dev_data = get_dev_data(dev);
	int error;

printk("0 ");
    error = configure(dev, spi_cfg);
	if (error == 0) {
		dev_data->busy = true;

		spi_context_buffers_setup(&dev_data->ctx, tx_bufs, rx_bufs, 1);
		spi_context_cs_control(&dev_data->ctx, true);

		transfer_next_chunk(dev);

printk("6 ");
        error = spi_context_wait_for_completion(&dev_data->ctx);
printk("7\r\n");
	}

	spi_context_release(&dev_data->ctx, error);

	return error;
}

static void transfer_next_chunk(struct device *dev)
{
	struct spi_nrfx_data *dev_data = get_dev_data(dev);
	const struct spi_nrfx_config *dev_config = get_dev_config(dev);
	struct spi_context *ctx = &dev_data->ctx;
	int error = 0;

printk("1 ");

	size_t chunk_len = spi_context_longest_current_buf(ctx);

	if (chunk_len > 0) {
printk("2 ");
		nrfx_spim_xfer_desc_t xfer;
		nrfx_err_t result;
		const u8_t *tx_buf = ctx->tx_buf;
#if (CONFIG_SPI_NRFX_RAM_BUFFER_SIZE > 0)
		if (spi_context_tx_buf_on(ctx) && !nrfx_is_in_ram(tx_buf)) {
			if (chunk_len > sizeof(dev_data->buffer)) {
				chunk_len = sizeof(dev_data->buffer);
			}

			memcpy(dev_data->buffer, tx_buf, chunk_len);
			tx_buf = dev_data->buffer;
		}
#endif
		if (chunk_len > dev_config->max_chunk_len) {
			chunk_len = dev_config->max_chunk_len;
		}

		dev_data->chunk_len = chunk_len;

		xfer.p_tx_buffer = tx_buf;
		xfer.tx_length   = spi_context_tx_buf_on(ctx) ? chunk_len : 0;
		xfer.p_rx_buffer = ctx->rx_buf;
		xfer.rx_length   = spi_context_rx_buf_on(ctx) ? chunk_len : 0;
		result = nrfx_spim_xfer(&dev_config->spim, &xfer, 0);
printk("3 ");
		if (result == NRFX_SUCCESS) {
			return;
		}

printk("* ");
		error = -EIO;
	}

	spi_context_cs_control(ctx, false);

	SYS_LOG_DBG("Transaction finished with status %d", error);

printk("4 ");
	spi_context_complete(ctx, error);
	dev_data->busy = false;
}

static inline void spi_context_complete(struct spi_context *ctx, int status)
{
#ifdef CONFIG_SPI_ASYNC
	if (!ctx->asynchronous) {
		ctx->sync_status = status;
		k_sem_give(&ctx->sync);
	} else {
		if (ctx->signal) {
#ifdef CONFIG_SPI_SLAVE
			if (spi_context_is_slave(ctx) && !status) {
				/* Let's update the status so it tells
				 * about number of received frames.
				 */
				status = ctx->recv_frames;
			}
#endif /* CONFIG_SPI_SLAVE */
			k_poll_signal(ctx->signal, status);
		}

		if (!(ctx->config->operation & SPI_LOCK_ON)) {
			k_sem_give(&ctx->lock);
		}
	}
#else
	k_sem_give(&ctx->sync);
printk("5 ");
#endif /* CONFIG_SPI_ASYNC */
}

DeclanTraill, 2019-01-17:
Sorry for the formatting of the pasted code (above) - The insert code function doesn't seem to preserve the code format.

nwsetec, 2019-01-17:
In the image below:
- The green line is toggled just before `k_sem_give(&ctx->sync);` in the `spi_context_complete` function.
- The purple line is toggled on each successful call to `nrfx_spim_xfer` in the `transfer_next_chunk` function (toggles 1, 2 and 3) and in the `spi_context_release` before the `k_sem_give(&ctx->lock);` (toggle 4).

In the first transfer it successfully makes it to k_sem_give (the 4th toggle) in the second it doesn't.

While gathering this data sometimes the thread the SPIM transfer was in would keep going and the Bluetooth peripheral function of the system would stop working and other times the reverse.

[image]

@anangl is this enough to go on?

anangl, 2019-01-17:
> @anangl is this enough to go on?

I think so. Thanks for providing so detailed reports.
Apologies for not being able to take a look at it earlier. I had to finish some other stuff. Now I can switch to this one and investigate a bit.

nwsetec, 2019-01-17:
@anangl cheers greatly appreciated!

DeclanTraill, 2019-01-18:
Some more Testing and information (plus test code) on Two cases of failure involving the ctx->sync semaphore:

----
Failed Test 1:

z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zzzzzzzzzzz0 1 2 3 6 .adef(4) 01 2 3 
zef(4) 0
1 2 3 (4) 0
zef1 4 5 ef(4) 0
z7 8
zzzzzzzzzzzzzzzz0 1 2 3 6 .adef(4) 0
1 2 3 ef1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zz0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zz0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zzzz0 1 2 3 6 .adef1 2 3 1 2 3 (4) 4 5  0
zef(4) 0
z7 8
zz0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zzzzzzzzzzz0 1 2 3 6 .adef(4) 0 2 3 
zef(4) 0
1 2 3 (4) 0
zef1 4 5 (4) 0
zefef(4) 0
zef(4) 0
ef(4) 0
zef(4) 0
ef(4) 0
zef(4) 0

In this case, the code stopped after the return (the 'z' printed) in function _impl_k_sem_take():

	int ret = _pend_current_thread(key, &sem->wait_q, timeout);
	printk("z");

	return ret;

But the code following on from the return (the print of '7') never executed:

in spi_context_wait_for_completion():

if (dbg_on)
	k_sem_take(&ctx->sync, K_FOREVER);
if (dbg_on) printk("7 ");

----
Failed Test 2:

z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
zzzz0 1 2 3 6 .adef1 2 3 (4)  2 3 0
zef(4) 0
 4 5 (4) 0
z7 8
zz0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
z0 1 2 3 6 .adef1 2 3 1 2 3 1 4 5 (4) 0
z7 8
1 2 3 zzzz0 1 2 3 6 .adef(4) 0
      zef(4) 0
1 2 3 (4) 0
zef1 4 5

In this case, the code stopped in _pend_current_thread() on the line:
unsigned int ret = _Swap(key);

----

The Test Code:

in sem.c

int _impl_k_sem_take(struct k_sem *sem, s32_t timeout)
{
	__ASSERT(!_is_in_isr() || timeout == K_NO_WAIT, "");
if (sem == sync_ptr) printk(".");
	sys_trace_void(SYS_TRACE_ID_SEMA_TAKE);
	unsigned int key = irq_lock();

	if (dbg_on)	if (sem == sync_ptr) printk("a");
	if (likely(sem->count > 0)) {
		if (dbg_on) if (sem == sync_ptr) printk("b");
		sem->count--;
		irq_unlock(key);
		sys_trace_end_call(SYS_TRACE_ID_SEMA_TAKE);
		if (dbg_on) if (sem == sync_ptr) printk("c");
		return 0;
	}

	if (timeout == K_NO_WAIT) {
		irq_unlock(key);
		sys_trace_end_call(SYS_TRACE_ID_SEMA_TAKE);
		return -EBUSY;
	}

	sys_trace_end_call(SYS_TRACE_ID_SEMA_TAKE);

	if (dbg_on) if (sem == sync_ptr) printk("d");
	my_key = key;
	int ret = _pend_current_thread(key, &sem->wait_q, timeout);
	printk("z");

	return ret;
}

in kswap.h

extern unsigned int my_key;
extern bool in_transceive;
extern int cc;
static inline unsigned int _Swap(unsigned int key)
{
	unsigned int ret;
	cc = 0;
	if (1) if (in_transceive && (my_key == key)) cc++;
	_check_stack_sentinel();
	if (1) if (in_transceive && (my_key == key)) cc++;
	_update_time_slice_before_swap();
	if (1) if (in_transceive && (my_key == key)) cc++;

#ifdef CONFIG_TRACING
	sys_trace_thread_switched_out();
#endif
	ret = __swap(key);
	if (1) if (in_transceive && (my_key == key)) cc++;
#ifdef CONFIG_TRACING
	sys_trace_thread_switched_in();
#endif

	return ret;
}

in sched.c

int _pend_current_thread(int key, _wait_q_t *wait_q, s32_t timeout)
{
	if (dbg_on) if (in_transceive && (my_key == key)) printk("e");
	pend(_current, wait_q, timeout);
	if (dbg_on) if (in_transceive && (my_key == key)) printk("f");
	unsigned int ret = _Swap(key);
	if (dbg_on) if (in_transceive && (my_key == key)) printk("(%d) %u\r\n", cc, ret);
	return ret;
}

in spi_context.h

static inline int spi_context_wait_for_completion(struct spi_context *ctx)
{
	int status = 0;
#ifdef CONFIG_SPI_ASYNC
	if (!ctx->asynchronous) {
		k_sem_take(&ctx->sync, K_FOREVER);
		status = ctx->sync_status;
	}
#else
	k_sem_take(&ctx->sync, K_FOREVER);
    printk("7 ");
    status = ctx->sync_status;
#endif /* CONFIG_SPI_ASYNC */

#ifdef CONFIG_SPI_SLAVE
	if (spi_context_is_slave(ctx) && !status) {
		return ctx->recv_frames;
	}
#endif /* CONFIG_SPI_SLAVE */

	return status;
}

anangl, 2019-01-21:
I analyzed the information you provided. But I haven't discovered anything new so far. I haven't tried to replicate the issue yet since I had only a couple of hours to spend on this. From your reports it seems that the hardware behaves as expected and the driver handles it correctly, but some strange problem occurs occasionally with waking up the thread the driver works in. Well, this you've found already yourself. The same pattern with using the sync semaphore is used in more SPI drivers (spi_context is common), so it is not something Nordic specific. Does the problem occur also when Bluetooth is disabled?
All this week I am on vacation so I won't be able to dive into it again till next Monday. Maybe some kernel expert could look at this in the meantime. @pizi-nordic Could you take a look?

pizi-nordic, 2019-01-21:
Hi. 

I will try to trace this issue tomorrow.
But just to rule-out obvious things: Have you verified that you are not running out of stack?

nwsetec, 2019-01-21:
Thanks  @anangl.

Hi @pizi-nordic, thanks for looking into this also.  I've looked at the each thread stack usage and there doesn't seem to be an issue there. I've configured CONFIG_STACK_SENTINEL=y and CONFIG_STACK_CANARIES=y and neither of these detect failures while running.  Would that be enough to verify there is no stack issue?

pizi-nordic, 2019-01-22:
@nwsetec: It depends on the stack usage pattern. For example if you have a kind of array on stack and you are not filling this array as a whole but use only a small fragment, you are likely to miss stack corruption as you can perform access over canary without modifying it. I recommend turning on MPU-based stack guard.

nwsetec, 2019-01-23:
@pizi-nordic ok thanks for this advice, I will try it.

nwsetec, 2019-01-28:
@pizi-nordic I've had the system running with CONFIG_MPU_STACK_GUARD=y and I'm seeing resets but no mention of the MPU.  When the reset occurs nothing is logged.

roland2000, 2019-02-22:
This Roland and I'm working with nwsetec.
I have investigated into this in the past few days and believe I found the root cause: SPI priority mis-configuration.
In our project we have the following Zephyr configs,
CONFIG_CPU_CORTEX_M_HAS_PROGRAMMABLE_FAULT_PRIOS = 1
CONFIG_ZERO_LATENCY_IRQS = undefined
And therefore, _IRQ_PRIO_OFFSET = 1

_IRQ_PRIO_OFFSET will be added to the priority setting configured in application, and as mentioned in the original post, we have CONFIG_SPI_0_IRQ_PRI=7.
After taking the _IRQ_PRIO_OFFSET, level 8 is passed into the _irq_priority_set(). However, the ARM CM4 has a 3-bit priority, so 8 overflows and becomes 0. 
This should have been caught by the ASSERT inside _irq_priority_set() but because we turned the ASSERT on and off during the development and happened to leave it off during that period of time.

I tested the system with CONFIG_SPI_0_IRQ_PRI set to 2 and 6. Both ran for about 17 hours with no crash, more testing is to be done.

So apparently if application uses interrupt priority 0 it will cause some kind of conflicts and result in system crash. 
My question is, is the conflict related to the kernel or something else, e.g., SoftDevice? My understanding is that if SoftDevice is not enabled the priority levels it reserves will be available to application. Please correct me if I'm wrong.
If the conflict is indeed with the kernel, may I know exactly what conflict it is and how it triggers the reset?

Below is some information I found online about priority levels occupied by SoftDevice.
http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.sds%2Fdita%2Fsoftdevices%2Fs130%2Fprocessor_avail_interrupt_latency%2Fexception_mgmt_sd.html

_The nRF52 SoC has eight configurable interrupt priorities ranging from 0 to 7 (with 0 being highest priority). On reset, all interrupts are configured with the highest priority (0).

**The SoftDevice reserves and uses the following priority levels, which must remain unused by the application programmer:**

Level 0 is used for the SoftDevice's timing critical processing.
Level 1 is used for handling the memory isolation and run time protection, see Memory isolation and runtime protection.
Level 4 is used by higher-level deferrable tasks and the API functions executed as SVC interrupts._

nwsetec, 2019-03-06:
In summary:
We had incorrectly configured the SPIM with CONFIG_SPI_0_IRQ_PRI=7, as when used with the _irq_priority_set() function, the priority used in the hardware register had been rolled over to be 0.  Note that building with CONFIG_ASSERT=y would have alerted us to this issue earlier with this output:
```
ASSERTION FAIL [prio <= ((1 << 3) - 1)] @ zephyr/arch/arm/core/irq_manage.c:114:

        invalid priority 7! values must be less than 7
```
The after setting CONFIG_SPI_0_IRQ_PRI=6 the SPIM driver no longer hangs.

closed 2019-03-06 by nwsetec