zephyrproject-rtos/zephyr #12454: doc: Some board images are pretty big (> 5MB)

chrta, 2019-01-13:
When i viewed the board documentation for https://docs.zephyrproject.org/latest/boards/arm/mimxrt1020_evk/doc/mimxrt1020_evk.html today on a slow internet connection, i noticed that loading the board image took a view seconds. 

When executing: `find boards -type f  -exec du -h {} + | sort -r -h | less` i noticed other large images:

~~~
8,9M    boards/arm/mimxrt1020_evk/doc/mimxrt1020_evk.jpg
4,2M    boards/arm/mimxrt1060_evk/doc/mimxrt1060_evk.jpg
3,0M    boards/arm/frdm_k64f/doc/frdm_k64f.jpg
2,7M    boards/shields/x_nucleo_iks01a2/doc/img/x-nucleo-iks01a2.jpg
...
~~~

Maybe it would make sense to shrink them since they are scaled down in the browser anyways.

nashif, 2019-01-18:
@dbkinder should make this part of the documentation guidelines. We also should go over all those images and standardise them, i.e. request png with a max size and resolution.

dbkinder, 2019-01-18:
I'm on it... all the big images are indeed in the boards area:
```
8.9M    ./boards/arm/mimxrt1020_evk/doc/mimxrt1020_evk.jpg
4.2M    ./boards/arm/mimxrt1060_evk/doc/mimxrt1060_evk.jpg
3.0M    ./boards/arm/frdm_k64f/doc/frdm_k64f.jpg
2.7M    ./boards/shields/x_nucleo_iks01a2/doc/img/x-nucleo-iks01a2.jpg
1.5M    ./boards/arm/olimex_stm32_e407/doc/img/olimex-stm32-e407-front.png
1.5M    ./boards/arm/disco_l475_iot1/doc/img/disco_l475_iot1.jpg
1.2M    ./boards/shields/x_nucleo_iks01a1/doc/img/x-nucleo-iks01a1.jpg
1.1M    ./boards/arm/96b_carbon/doc/img/96b-carbon-front.png
984K    ./boards/arm/nrf52_adafruit_feather/doc/img/nRF52Feather.png
904K    ./boards/arm/colibri_imx7d_m4/doc/colibri_imx7d.png
828K    ./boards/arm/reel_board/doc/img/reel_board_debug.png
816K    ./boards/arm/cy8ckit_062_wifi_bt_m4/doc/img/cy8ckit_062_wifi_bt.png
816K    ./boards/arm/cy8ckit_062_wifi_bt_m0/doc/img/cy8ckit_062_wifi_bt.png
788K    ./boards/arm/reel_board/doc/img/reel_board_tp.png
764K    ./boards/arm/reel_board/doc/img/reel_board_descr_back.png
764K    ./boards/arm/nucleo_f411re/doc/img/nucleo64_perf_logo_1024.png
764K    ./boards/arm/nucleo_f401re/doc/img/nucleo64_perf_logo_1024.png
756K    ./boards/arm/mps2_an385/doc/img/mps2.png
744K    ./boards/arm/udoo_neo_full_m4/doc/udoo_neo_full.jpg
668K    ./boards/arm/sam_e70_xplained/doc/img/sam_e70_xplained.png
584K    ./boards/arm/arduino_due/doc/img/arduino_due.png
568K    ./boards/arm/reel_board/doc/img/reel_board_descr_front.png
564K    ./boards/arm/nucleo_f401re/doc/img/nucleo_f401re_arduino.png
556K    ./boards/arm/nucleo_f411re/doc/img/nucleo_f411re_arduino.png
552K    ./boards/arm/nucleo_l476rg/doc/img/nucleo_l476rg_arduino.png
512K    ./boards/arm/nucleo_l432kc/doc/img/nucleo_l432kc_arduino_nano.png
500K    ./boards/arm/nucleo_f412zg/doc/img/Nucleo144_perf_logo_1024.png
500K    ./boards/arm/nucleo_f207zg/doc/img/Nucleo144_perf_logo_1024.png
496K    ./boards/arm/nucleo_f413zh/doc/img/nucleo_f413zh_zio_left.png
496K    ./boards/arm/hexiwear_k64/doc/hexiwear_k64.jpg
492K    ./boards/arm/nucleo_f412zg/doc/img/nucleo_f412zg_zio_left.png
...
```

galak, 2019-01-18:
I've used this site to reduce image file size:

https://imagecompressor.com/

PR #12593 (2019-01-19): doc: reduce image size of board pics
Some board images were unnecessarily large and slowing down the browser

Fixes #12454
closed 2019-01-19 by nashif