zephyrproject-rtos/zephyr #12455: Fatal fault when openthread commissioner starts

arn-soft, 2019-01-13:
**Describe the bug**
Openthread commands works well. But we run **ot commissioner start** command on uart, it throws fatal fault.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build && cd build
2. cmake -GNinja -DBOARD=nrf52840_pca10056 -DOVERLAY_CONFIG=overlay-ot.conf ..
3. ninja

**Expected behavior**
It should start commissioner.

**Impact**
Device can't be commissioned.

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment:**
OS: Linux
Toolchain: Zephyr-sdk
Commit id : c885cb5

jukkar, 2019-01-15:
@rlubos have you seen this issue before?

rlubos, 2019-01-15:
@jukkar Nope, I haven't seent this before. I'll investigate once I'm done with current task.

rlubos, 2019-01-18:
Apparently it's not exactly a bug, it's a stack overflow issue when commissioner is being enabled. I'll submit a PR that increse default stack size for OT when commisioner is on.