zephyrproject-rtos/zephyr #12462: my nrf52 DK gets damaged partially after hundreds of erase and write?

summer20100514, 2019-01-14:
I choose one NRF52 DK(nrf52840_pca10056) for my daily development. After hundreds of erase and write cycles, the DK won't behaves like it should do. The other DKs runs well while this one not. But some basic example runs well. How can I figure out if this DK is damaged? If so, why is it damaged partially?

what I have tested is, zephyr ibeacon sample works well, while peripheral sample wont work as expected. However, other DKs work well.

peripheral sample output

> ***** Booting Zephyr OS zephyr-v1.13.0-3079-g4e23d6c875 *****
> Bluetooth initialized
> [00:00:01.514

and it is stuck at 
static inline void cpu_sleep(void)
{
  __WFE();
  _**_SEV(); // it stucks here**
  __WFE();
}

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain None
 - Commit SHA c922ce6ce45173c2b1f2c8c48f57f275d9a046d8

summer20100514, 2019-01-14:
I also write a post on nordic devzone, but I am not sure whether if it is caused by hardware failure or by software damage.

carlescufi, 2019-01-14:
This might be a hardware issue, but I've copied a couple of people to see if they can provide additional information

summer20100514, 2019-01-14:
@carlescufi Thank you. What still confused me is the chip shouldn't be damaged by just hundreds of erase cycles.

closed 2019-01-16 by summer20100514