zephyrproject-rtos/zephyr #12478: tests/drivers/ipm/peripheral.mailbox failing sporadically on qemu_x86_64 (timeout)

nashif, 2019-01-15:
see https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/31042/6/tests, the test times out with

```
***** Booting Zephyr OS zephyr-v1.13.0-3219-gdfc8f0a6d5 *****
starting test - Test IPM
ipm_recv0: 'everything is awesome'
ipm_recv0: 'Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor '

```

andyross, 2019-01-15:
OK, confirmed.  I got 3 failures out of 10 tries running this test under load on my CI (-ish) docker instance.  Interestingly not one failure in 40 running the same binaries built with the same SDK under the host qemu.

So... plausibly timing related.  Could be as simple as a stack overflow that depends on when the interrupt lands, if I'm lucky.

wentongwu, 2019-03-14:
tried so many times with script to reproduce this issue, but failed to reproduce it on my Ubuntu18.04 PC so far.  talked with @andyross, I will try to reproduce it with docker. If anyone can easily reproduce it, please tell me... thanks

wentongwu, 2019-03-14:
with https://docs.docker.com/install/linux/docker-ce/ubuntu/, didn't setup docker successfully(welcome setup docker's BKM). So try to reproduce it with Cinly's Fedora PC(Commit SHA: g4278bf567348).

test it with below command  for i in {1..100}; do scripts/sanitycheck -T tests/drivers/ipm/ -p qemu_x86_64; done (disable the exclude #platform_exclude: qemu_x86_64 in testcase.yaml)

when open SMP in soc config, this issue can reproduce. with SMP disabled, no fail occurred. So I'm checking SMP's code...

wentongwu, 2019-03-26:
this issue has already submitted PR(#14619 ), but this PR depends on https://github.com/zephyrproject-rtos/zephyr/issues/14807.

andyross, 2019-04-04:
Took a look here.  As far as I can tell, the earlier problem has disappeared and the only issue is, as @wentongwu discovered, SMP correctness.  There's no point in trying to finesse that with a controversial patch at this stage.  We have **lots** of known SMP safety issues in the test code already, and the workaround is just to disable SMP on those tests.  We can come back in 1.15 as part of the work for #14639 and fix things at our leisure.

nashif, 2019-04-28:
problem still exist

```
c[?7l[2J[0mSeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.14.0-3-g9a8467d83aaf *****
starting test - Test IPM
ipm_recv0: 'everything is awesome'
ipm_recv0: 'Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor '
ipm_recv0: 'incididunt ut labore et dolore magna aliqua. Ut enim ad minim veniam, quis nost'
ipm_recv0:ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../../../include/spinlock.h:78
	Recursive spinlock
*** FATAL ERROR vector 6661 code 1016
***  RIP 9:0x100e46 RSP 16:0x105ba8 RFLAGS 0x2
***  RAX 0x105327 RCX 0xa RDX 0x3f8 RSI 0x4e RDI 0x1052d9
***  R8 0xffffffff R9 0x105ac8 R10 0x0 R11 0x0
```

rljordan, 2019-08-13:
Can you please retest this?  Might be fixed.

wentongwu, 2019-08-14:
With 12 hours stress test, can’t reproduce.

rljordan, 2019-08-14:
Anas can we close this?

jenmwms, 2019-08-23:
Please note that if we close this, we need to revisit the commit 0dd0f9f52a10f1bee71e21ccc7daba7c59b89dbf to enable (it was disabled in the commit to allow testing).

wentongwu, 2019-08-29:
The case is really not designed for SMP as PR #14619 found, and I will disable SMP for this case, but I will do the change as commit https://github.com/zephyrproject-rtos/zephyr/pull/18531/commits/ab3a20921d11392be911861d448126cae72da614 do.

nashif, 2019-08-29:
test not supporting SMP is not a medium bug, lowering priority

pfalcon, 2020-04-15:
Hello again. We have tests/drivers/ipm/ failing reproducibly as part of https://github.com/zephyrproject-rtos/zephyr/pull/23693 , with a stack overflow. I proceeded to check if bumping stack would resolve it. Of course, I started with trying that with master (7ca6a05159e9dfdce3e257ec4fc4d5b5b7ad3127) first (Zephyr SDK 0.11.2). 

I found that 3 out of 3 initial run hanged with:
~~~
[QEMU] CPU: qemu64,+x2apic
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 8
*** Booting Zephyr OS build zephyr-v2.2.0-1489-g7ca6a05159e9  ***
starting test - Test IPM
ipm_recv0: 'everything is awesome'
ipm_recv0: 'Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmod tempor '
ipm_recv0: 'incididunt ut labore et dolore magna aliqua. Ut enim ad minim veniam, quis nost'
QEMU: Terminated
~~~

Then, in next runs, can't reproduce any more. So, initially reopened, but closing again.

PR #15191 (2019-04-05): tests/drivers/ipm: Enable on (uniprocessor) x86_64
[Let's just skip the SMP work in #14619 and commit the obvious workaround for now.   There's a giant task for 1.15 to make all the tests and subsystems SMP-safe anyway, there's no need to jump through different hoops for IPM of all things.]

This test was having trouble earlier (at least with some toolchains),
but whatever was causing that seems to have been fixed.  The only
remaining issue is that neither the IPM console code nor the test are
SMP-safe.  There's a bug (#14639) tracking the need to get these
working, but for now the straightfor [... cut]
commit 0c594714 (2019-04-05): tests/drivers/ipm: Enable on (uniprocessor) x86_64

This test was having trouble earlier (at least with some toolchains),
but whatever was causing that seems to have been fixed.  The only
remaining issue is that neither the IPM console code nor the test are
SMP-safe.  There's a bug (#14639) tracking the need to get these
working, but for now the straightforward workaround is just to disable
SMP.

And even long term, IPM is an oddball interprocessor communication
mechanism designed for asymmetric multiprocessing devices like Quark
SE and doesn't seem like an obvious fit for a SMP machine.

Fixe [... cut]
PR #18767 (2019-08-29): tests/drivers/ipm: Enable on x86_64
this case isn't designed for SMP as PR #14619 found, setting CONFIG_MP_NUM_CPUS=1
as commit https://github.com/zephyrproject-rtos/zephyr/pull/18531/commits/ab3a20921d11392be911861d448126cae72da614 did to run the full SMP kernel config, but with only one CPU available.

I run the stress test for this case with this PR, I never see a failure.

Fixes: #12478.
commit a58d730d (2019-08-29): tests/drivers/ipm: Enable on x86_64

this case isn't designed for SMP, setting CONFIG_MP_NUM_CPUS=1 to
run the full SMP kernel config, but with only one CPU available.

Fixes: #12478.
closed 2020-04-15 by pfalcon