zephyrproject-rtos/zephyr #12483: [Bluetooth Mesh] Is the BLE Mesh stack certificated with SIG Profile Tuning Suite tool?

gofortime, 2019-01-15:
Hi experts,

I am wondering if the BLE Mesh stack was ever certificated/tested with the PTS (Profile Tuning Suite) tool from SIG (more details about PTS can be found in https://www.bluetooth.com/develop-with-bluetooth/qualification-listing)? I tried the find the test report from Zephyr website but fail. If PTS certificated, can anyone tell me where I can find the report and which sample I can build to test against PTS? For example, I found some mesh samples in "samples/boards/nrf52/mesh/" directory of zephyr project. But I am not sure whether they can be used for PTS test or not.

Can anyone help? Really appreciate.

Thanks,
Vincent.

jhedberg, 2019-01-15:
The stack gets run periodically (about once a week) against the PTS. There does not exist a qualification listing for it yet however. The sample application that's used for this automated testing is tests/bluetooth/tester and it is used together with the AutoPTS framework: https://github.com/intel/auto-pts

jhedberg, 2019-01-15:
The set of profiles that are supported (you'll see Mesh is included): https://github.com/intel/auto-pts/tree/master/ptsprojects/zephyr

gofortime, 2019-01-16:
@jhedberg Thanks a lot for the information. Let me do an investigation.

closed 2019-01-16 by gofortime