zephyrproject-rtos/zephyr #12488: RedBear Nano v2 Mesh Instruction Fault

DrPresident, 2019-01-15:
I am attempting to get Bluetooth Mesh working on the RedBear BLE Nano v2, however I get instruction faults when running any of the sample applications that build. I have successful builds of non-bluetooth applications, and I have attempted on a few different boards of the same model. I am building on Debian stretch with ninja, and deploying by copy-pasting the .bin or the .hex file into the mounted DAPLINK board. 

Board:
BLE Nano 2
Nordic nRF52832

DAPLink Firmware
```
Unique ID: 10930000313878511239333437353533a5a5a5a597969906
HIC ID: 97969906
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0002
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Interface CRC: 0xdce4665e
Remount count: 0
```

zephyr/samples/bluetooth/mesh_demo

```
cmake -GNinja -DBOARD=nrf52_blenano2 ..
ninja
cp zephyr/zephyr.bin /tmp/DAPLINK/
```

Serial Output
```
***** Booting Zephyr OS zephyr-v1.13.0-2588-gee0b61fff8 *****
Initializing...
Unicast address: 0x0b0c
***** Kernel Panic! *****
Current thread ID = 0x20001210
Faulting instruction address = 0x2510

```
addr2line -e build/zephyr/zephyr.elf 0x2510
`/home/drpresident/zephyr/subsys/settings/src/settings_init.c:68`

settings_init.c
```
static void settings_init_fcb(void){
    ...
    config_init_settings_fcb.cf_fcb.f_sector_cnt = cnt; // line 68
    ...
}
```

carlescufi, 2019-01-15:
Adding @jhedberg and @nvlsianpu since this seems like a combination of mesh with the settings subsystem

Laczen, 2019-01-17:
@DrPresident, I think there is an error in the dts for the Redbear Nano v2: the storage area is defined as reg = <0x0007D000 0x00004000>, this is a to large area, the length should be smaller than 0x00003000.
There is also an option CONFIG_MPU_ALLOW_FLASH_WRITE=y that seems to be needed sometimes to allow flash writes.

Laczen, 2019-01-17:
@DrPresident, I would copy the flash configuration from nrf52_pca10040:
scratch: reg = <0x00070000 0xa000>
storage: reg = <0x0007a000 0x6000>
I am not sure, but I think the storage area for settings in fcb is divided into spaces (default 8) and these have to be a multiple of the flash page size, with a size of 0x3000 this would not be true. Increase the size to 0x6000 and each section becomes a multiple of the page size.

DrPresident, 2019-01-21:
I applied the scratch and storage register changes, and I received the same errors with and without the CONFIG_MPU_ALLOW_FLASH_WRITE=y in the prj.conf file

Laczen, 2019-01-21:
Hi @DrPresident,
I tried using a nrf52_pca10040 as a redbear nano v2. After changing the serial port to the correct numbers everything worked directly. Could it be a problem with the daplink programming ? Can you use the daplink also as a debugger ? If so, could you try erasing (after upload) the flash area starting at 0x7a000 size 0x6000, and after erase do a reset. I think you should be able to use the daplink with openocd (the standard runner for ble_nano v2), but since I don't have a daplink or a ble nano v2 I cannot test this.

DrPresident, 2019-01-21:
When I use nrf52_pca10040 it appears to run just fine. I will work with that for now thank you!

nvlsianpu, 2019-01-22:
So storage area size is ok in blenano2 - settings will consume as much it can, up to the limit stands by SETTINGS_FCB_NUM_AREAS.

closed 2019-01-22 by nvlsianpu