zephyrproject-rtos/zephyr #12490: Produced ELF does not follow the linux ELF spec

sigvartmh, 2019-01-15:
**Describe the bug**
Tried to ask this on slack but got no answer. I've a question about the output of the linker-script and elf for ARM. Why does it not follow the Linux ELF spec/convention for the section headers when it comes to text, bss etc. but does so for `.comment`?

[SPEC](http://refspecs.linuxbase.org/elf/elf.pdf ) As the output of the section headers in the ELF produced does not follow the spec and also outputs a _TEXT_SECTION_NAME_2 section header which I assume is not supposed to be there.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=<an ARM board>
3. make
4. `readelf -S main <builddir>/zephyr/zephyr.elf

**Expected behavior**
To follow the ELF spec and by then having a `.` before the section names described in `include/linker/sections.h`

**Impact**
Mostly annoyance but also makes including other parts into the linker-script with `custom-<type>.ld` a bit harder.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
This is an open question I'll submit a PR with a fix for the sections named specified by the SPEC but looking at the project this could be a design decision so I'm mostly looking for answeres.

SebastianBoe, 2019-01-16:
> will this still fail for XiP?

I don't know. Is there anything in the git history that can help us here?

sigvartmh, 2019-01-16:
> > will this still fail for XiP?
> 
> I don't know. Is there anything in the git history that can help us here?

Can't find any decent git history for it. As far back as I got https://github.com/zephyrproject-rtos/zephyr/commit/8ddf82cf70dc6f951ab477f325dee0efde3ec589#diff-f48aac91257b5dee83fa3cb9400edcbe
it seems to have been there from the First commit.

SebastianBoe, 2019-01-16:
Then I think it is reasonable to assume the comment is invalid.

sigvartmh, 2019-01-16:
Then the only thing I'm concerned about is that the sanitycheck will not be correct https://github.com/zephyrproject-rtos/zephyr/blob/81ec05548dbafb602d2d400ef86ffcf2e1732c31/scripts/sanitycheck#L666

SebastianBoe, 2019-01-16:
Right. All in-tree references to the old section names must be ported to the new section names.

andrewboie, 2019-01-17:
What's the specific issue that needs to be fixed here?
I don't believe there's any requirement that Zephyr follow Linux section naming conventions

sigvartmh, 2019-01-17:
> What's the specific issue that needs to be fixed here?
> I don't believe there's any requirement that Zephyr follow Linux section naming conventions

It's not really an issue, it's just nice to have something to base the section naming on. It's also would help on-boarding linux developers I guess?

That being said the real minor issue is probably `_TEXT_SECTION_NAME_2` which is a bit of a weird section name.

However I think it does no harm? and if you think it's not worth changing I'm not reallt opposed to it either. It is however a bit easier to put your stuff in to the correct section with `custom-sections.ld` files if the naming convention is consistent and you have some documentation to look at. At least that's my argumenet.

Another for the `_TEXT_SECTION_NAME_2` issue is:

>pre-existing linker script code that places code in the .text output
>section should, when integrated with Zephyr, place code after
>_vector_end. Not before.

andrewboie, 2019-01-17:
_TEXT_SECTION_NAME_2 is definitely a bug, I'll look into it

sigvartmh, 2019-01-18:
I've fixed it here https://github.com/zephyrproject-rtos/zephyr/pull/12510 in https://github.com/zephyrproject-rtos/zephyr/pull/12510/commits/83878df652f774440275a444263b64107ab50dd3 but maybe one want to rename the section https://github.com/zephyrproject-rtos/zephyr/pull/12510 to something like `init` /`.init` and have the actual text symbols be put in `text/.text` output section?

SebastianBoe, 2019-01-30:
> I think it may be possible to roll all or most of the current special sections we have into the output .data section

This simplification would be great to see (if possible).

SebastianBoe, 2019-08-14:
The question has been answered so I want to label this as an enhancement.

closed 2020-02-13 by nashif