zephyrproject-rtos/zephyr #12501: nRF52: UARTE lacks pm interface

tautologyclub, 2019-01-15:
I noticed that `drivers/serial/uart_nrfx_uarte.c` doesn't initialize the power management callbacks, but it does for regular UART. However, when inspecting `drivers/serial/uart_nrfx_uart.c` more closely, you actually catch a bunch of hardcoded references to uart0, which leads me to believe that it's impossible to suspend uart1 under the current API. 

Am I missing something here?

tautologyclub, 2019-01-17:
So, if this is actually the case (which it seems like), in my mind it should be assigned very high priority because UART_1 is virtually unusable for nRF52 in its current state. I'd be happy to assist with the fix, but would like Nordic to comment first.

carlescufi, 2019-01-17:
@tautologyclub I think either @jarz-nordic or @pizi-nordic can comment on this

tautologyclub, 2019-01-17:
Looking at the HAL, it sort of looks like this would be very easy by using 

```__STATIC_INLINE void nrf_uarte_disable(NRF_UARTE_Type * p_reg)```

and getting p_reg from 

```get_uarte_instance(dev)```

from `drivers/serial//home/benjamin/repos/zephyr/drivers/serial/uart_nrfx_uarte.c`

Is this the general idea?

jarz-nordic, 2019-01-24:
@tautologyclub : UART and UARTE SHIM has been prepared by me at the same time. I had internall discussion about the approach and we decided that UARTE shim is multiinsance while UART SHIM is not.
The reason is quite simple: there is no and there will be no Nordic microcontroller with more than 1 UART instance.
And if by any chance this will change we will update the SHIM.

jarz-nordic, 2019-01-24:
Regarding power management for UARTE we will try to prepare a fix for that shortly.

tautologyclub, 2019-01-24:
@jarz-nordic Thanks, that's great. Are we talking days, weeks or months?

jarz-nordic, 2019-01-24:
@tautologyclub : Once new UART API is merged I hope to have it in next 2-3 weeks.

carlescufi, 2019-03-06:
@jarz-nordic any update on this?

carlescufi, 2019-03-26:
@tautologyclub https://github.com/zephyrproject-rtos/zephyr/pull/13064

PR #13064 (2019-03-28): drivers: nrf: Rework UARTE shim for uart ASYNC API and add power management
Rework uart_nrfx_uarte shim to work with asynchronous API.

Fix power management issues

Fixes #12501
closed 2019-03-28 by nashif