zephyrproject-rtos/zephyr #12504: STM32: add USB_OTG_HS example

Jeepgoing, 2019-01-16:
I am now use stm32f407igt* board with usb hs mode, anybody can give me some advise about how to config? thanks a lot.

I have add some code to my board.

on dts:
`
&usbotg_hs { 
    status = "ok";
};
`

on pinmux.c
`
#ifdef CONFIG_USB_DC_STM32 
{STM32_PIN_PA3, STM32F4_PINMUX_FUNC_PA3_OTG_HS_ULPI_D0},
{STM32_PIN_PB0, STM32F4_PINMUX_FUNC_PB0_OTG_HS_ULPI_D1},
{STM32_PIN_PB1, STM32F4_PINMUX_FUNC_PB1_OTG_HS_ULPI_D2},
{STM32_PIN_PB10, STM32F4_PINMUX_FUNC_PB10_OTG_HS_ULPI_D3},
{STM32_PIN_PB11, STM32F4_PINMUX_FUNC_PB11_OTG_HS_ULPI_D4},
{STM32_PIN_PB12, STM32F4_PINMUX_FUNC_PB12_OTG_HS_ULPI_D5},
{STM32_PIN_PB13, STM32F4_PINMUX_FUNC_PB13_OTG_HS_ULPI_D6},
{STM32_PIN_PB5, STM32F4_PINMUX_FUNC_PB5_OTG_HS_ULPI_D7},
{STM32_PIN_PH4, STM32F4_PINMUX_FUNC_PH4_OTG_HS_ULPI_NXT},
{STM32_PIN_PI11, STM32F4_PINMUX_FUNC_PI11_OTG_HS_ULPI_DIR},
{STM32_PIN_PC0, STM32F4_PINMUX_FUNC_PC0_OTG_HS_ULPI_STP}, 
{STM32_PIN_PA5, STM32F4_PINMUX_FUNC_PA5_OTG_HS_ULPI_CK},
#endif  /* CONFIG_USB_DC_STM32 */
`

erwango, 2019-02-15:
@Jeepgoing, do you still hae the issue?
@ydamigos , maybe you would have insights?

Jeepgoing, 2019-02-15:
@erwango yes, it's still can't work, if i have any progress i will show it.

erwango, 2019-02-15:
@metapsychologe , since you ported usb on stm32f4_disco, maybe you can have a check?

ydamigos, 2019-02-15:
@Jeepgoing Does your board have an external ULPI_PHY? Checking stm32f407igt reference manual I see that OTG_HS has an internal OTG_FS_PHY which uses pins:
```
STM32F4_PINMUX_FUNC_PB14_OTG_HS_DM
STM32F4_PINMUX_FUNC_PB15_OTG_HS_DP
```

Jeepgoing, 2019-02-16:
@ydamigos  yeap, my board have an external ULPI_PHY, so you mean i need use internal OTG_FS_PHY?

ydamigos, 2019-02-16:
> yeap, my board have an external ULPI_PHY, so you mean i need use internal OTG_FS_PHY?

The STM32 usb driver wasn't tested with an external ULPI PHY, as far as I know. Maybe it needs a couple tweaks to work with external PHYs. If you want to test it with full-speed usb you could use the internal  OTG_FS_PHY.

Jeepgoing, 2019-02-16:
@ydamigos  our product must use usb HS mode for transmit data, FS mode can't satisfy our request. and i have tested FS mode.

ydamigos, 2019-02-16:
@Jeepgoing I can only provide you with some pointers. I don't have the hardware to test OTGHS with external PHY. 

One thing that should change in the driver for external ULPI PHY is `usb_dc_stm32_state.pcd.Init.phy_itface`. It's value should be `USB_OTG_ULPI_PHY`.
Also, you should enable the clocks for OTGHSULPI:
```
LL_AHB1_GRP1_EnableClock(LL_AHB1_GRP1_PERIPH_OTGHSULPI);
LL_AHB1_GRP1_EnableClockLowPower(LL_AHB1_GRP1_PERIPH_OTGHSULPI);
```
Again, I don't know if tha's enough to get it work but you could give it a try.

Jeepgoing, 2019-02-18:
@ydamigos  ok, i will try

Jeepgoing, 2019-02-19:
@ydamigos  i have followed your advise,but it's can't working too. i will do more test when i have time.

Jeepgoing, 2019-07-01:
@ydamigos Hi, do you have plan to support `OTGHS`

ydamigos, 2019-07-01:
@Jeepgoing As far as I know, OTG HS works on STM32 boards with internal HS PHY (I don't have the hardware to test it my self).
@finikorg If I remember correctly, you have tested STM32 OTG in high speed, haven't you?

finikorg, 2019-07-02:
> @finikorg If I remember correctly, you have tested STM32 OTG in high speed, haven't you?

Yes, IIRC on `stm32f723e-disco`

Jeepgoing, 2019-07-03:
@finikorg Hi, can you show me your test code? or give me some advise about how to change my code based on master code.

finikorg, 2019-07-03:
> @finikorg Hi, can you show me your test code? or give me some advise about how to change my code based on master code.

You can use any sample with overlay like:
```
cat stm32f723e_disco.overlay
&usbotg_fs {
	status = "disabled";
};

&usbotg_hs {
	status = "ok";
};
```

Jeepgoing, 2019-07-03:
is it any board support external PHY now? if i only modify 

> You can use any sample with overlay like:
> 
> ```
> cat stm32f723e_disco.overlay
> &usbotg_fs {
> 	status = "disabled";
> };
> 
> &usbotg_hs {
> 	status = "ok";
> };
> ```

it's can't work.

finikorg, 2019-07-03:
@Jeepgoing this works for the board `stm32f723e_disco` only, this board has 2 USB sockets, do not know about other boards, probably should work the same way.

jfischer-phytec-iot, 2019-09-16:
I seems to be resolved, if not reopen.

closed 2019-09-16 by jfischer-phytec-iot

arbrauns, 2022-10-20:
Can someone reopen this? I don't think it was solved at all, there is still no example for using an external ULPI PHY with USB_OTG_HS.

erwango, 2022-10-20:
> Can someone reopen this? I don't think it was solved at all, there is still no example for using an external ULPI PHY with USB_OTG_HS.

Not sure re-opening 3 years old issues has an interest. If facing issue with external phy, please raise a dedicated issue.
Regarding the example: to provide an example of external phy usage, we need a hw available in tree. I'm not sure we have that today (correct me if I'm wrong).

XenuIsWatching, 2022-11-16:
I'm not even sure an external PHY is even working with the implementation within `usb_dc_stm32.c`. no where do I see the definition `USB_OTG_ULPI_PHY` set for `phy_itface`

jfischer-no, 2022-11-21:
> Can someone reopen this? I don't think it was solved at all, there is still no example for using an external ULPI PHY with USB_OTG_HS.

Please open a new issue or discussion and describe your problem. Please stop abusing closed issues.