zephyrproject-rtos/zephyr #12509: Fix rounding in _ms_to_ticks()

pizi-nordic, 2019-01-16:
At the moment the _ms_to_ticks() rounds down, which results in premature expiration of timeouts scheduled just before edge of tick. To fix that we have to revert https://github.com/zephyrproject-rtos/zephyr/pull/9745 in order to restore rounding up.

pabigot, 2019-01-16:
That's one place where things get rounded down.  The same truncation happens in `z_clock_elapsed` in multiple architectures where fractional ticks are discarded.  Overall the need to traffic in durations that are downsampled from the hardware clock resolution in multiple places makes proper management of sleeps and alarms quite difficult.

ioannisg, 2019-08-17:
@pizi-nordic @carlescufi is this still applicable, and if so, how can we address it ifor 2.0?

pizi-nordic, 2019-08-19:
IMHO it should be fixed. I can prepare PR after I finish userspace logging.

ioannisg, 2019-08-27:
> 
> 
> IMHO it should be fixed. I can prepare PR after I finish userspace logging.

Could you please address this for 2.0 @pizi-nordic ?

carlescufi, 2019-08-27:
> > IMHO it should be fixed. I can prepare PR after I finish userspace logging.
> 
> Could you please address this for 2.0 @pizi-nordic ?

Or actually, will this be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/17155?
And if so, can we just close this and track the issue directly in https://github.com/zephyrproject-rtos/zephyr/issues/9904 ?

pizi-nordic, 2019-08-28:
@carlescufi: Yes, it will be fixed in #17155. However I will prepare a PR with this because #17155 will not go into 2.0 and now the change is safe as #16782 is merged.

pabigot, 2019-10-08:
This is pending in #19591 which uses ceil for ms_to_ticks.

PR #19591 (2019-11-08): New, orthogonal and complete time conversion API
Before pushing a big refactoring of the organization of #17155, it seems prudent to extract the bits that weren't controversial and see if we can merge those in isolation.

First up is the time conversion API.  This is the code as it appears in the original, with two additions:

1. The automatic conversion generator is extended to emit nanosecond conversions, and the test rig is updated accordingly for coverage.  This is done as a separate commit for simplicity, but I can fold it in if people want.

2. The older conversion utilities have been de [... cut]
closed 2019-11-08 by carlescufi