zephyrproject-rtos/zephyr #12527: Create a WebUSB subsys driver

larsgk, 2019-01-16:
WebUSB enabled devices are able to communicate directly with web applications with browsers implementing the spec.  Currently, there is a sample app (slightly outdated and static features) under /samples/subsys/usb/webusb

It would make sense to create a real subsys WebUSB 'driver' that can easily be included in projects that need WebUSB support.  Often these will also have the need for other classes in parallel (resulting in a composite USB device description).

As a starting point, we can use the sample mentioned above with the following changes:

- [ ]  Functionality to change manufacturer and product string
- [ ] Functionality to change serial runtime, e.g. to assign unique IDs from hardware
- [ ] The possibility to merge with other interfaces (e.g. CDC & WebUSB)
- [ ] License fix -> Apache2 (check with Intel?)

Also, it would be a good idea to involve Reilly Grant ( @reillyeon ) and Kenneth Christiansen ( @kenchris ) in the process.

larsgk, 2019-01-16:
/cc @carlescufi

markus-becker-tridonic-com, 2020-01-29:
I would really like to use WebUSB, possibly in combination with ACM and/or ECM. Would this use-case be possible with the current WebUSB sample or does the use-case require this issue to be tackled? Seems no-one is working on this currently, or?

larsgk, 2020-01-29:
The sample also has some handshake issues on windows (it seems).  I ran a workshop using the nRF52 Dongle and the Web USB stuff worked fine on Android, Linux and OSX - but not Windows even though it had the BOS and more in place that *should* make it PnP (I did a similar solution a few years ago using ARM Mbed, which works on all OSes -> https://github.com/empirikit )

markus-becker-tridonic-com, 2020-01-29:
Ideally, I would like to have WebUSB combined with the shell: https://github.com/zephyrproject-rtos/zephyr/issues/20409, so that you would have the Zephyr shell available in a browser.

jfischer-phytec-iot, 2020-01-29:
> WebUSB enabled devices are able to communicate directly with web applications with browsers implementing the spec. Currently, there is a sample app (slightly outdated and static features) under [...]

No, I do not think there should be a "WebUSB 'driver'" and not in the subsys. Nothing prevents you to use other classes in your WebUSB application.

> As a starting point, we can use the sample mentioned above with the following changes:
> 
>     * [ ]   Functionality to change manufacturer and product string
> 

What does it have to do with the sample? You can change PID and VID.

>     * [ ]  Functionality to change serial runtime, e.g. to assign unique IDs from hardware

Per default Serial Number will be obtained via HWINFO subsystem. Application can/should not change it after enumeration.

>     * [ ]  The possibility to merge with other interfaces (e.g. CDC & WebUSB)

We have support for composite configuration. Probably WebUSB sample needs some rework to show this.

>     * [ ]  License fix -> Apache2 (check with Intel?)
> 
> 

Where is this license issue?
[SPDX-License-Identifier: Apache-2.0](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/subsys/usb/webusb/src/webusb.c#L4l)

larsgk, 2020-01-29:
@jfischer-phytec-iot - It's been a year since the issue was created. Some comments below:

> No, I do not think there should be a "WebUSB 'driver'" and not in the subsys. Nothing prevents you to use other classes in your WebUSB application.
I think it makes sense to at least have a common place for cross host OS compliance (Windows being the most difficult to get true PnP working) handshaking. That is the biggest pain when developing USB firmware for true PnP devices that are not part of the standard classes (HID, CDC, ...)

> > As a starting point, we can use the sample mentioned above with the following changes:
> > ```
> > * [ ]   Functionality to change manufacturer and product string
> > ```
> 
> What does it have to do with the sample? You can change PID and VID.

The strings are what appear to the OS in true PnP (not talking crippled windows *.inf files here ;)). At least a year ago, those were hard coded deep in the source.

> > ```
> > * [ ]  Functionality to change serial runtime, e.g. to assign unique IDs from hardware
> > ```
> 
> Per default Serial Number will be obtained via HWINFO subsystem. Application can/should not change it after enumeration.
> 

Not changing after enumeration but being able to dynamically set based on e.g. chip ID before handshake.
> We have support for composite configuration. Probably WebUSB sample needs some rework to show this.
>
>
> Where is this license issue?
> [SPDX-License-Identifier: [...]

AFAIK, A year ago it had some original Intel license on it (doesn't matter now)

henrikbrixandersen, 2024-11-19:
@larsgk https://github.com/zephyrproject-rtos/zephyr/pull/76326 may be of interest to you regarding this topic.

larsgk, 2024-11-19:
> @larsgk #76326 may be of interest to you regarding this topic.

Thanks @henrikbrixandersen - I think most of the original intention of the ticket has been solved in cleanups of the WebUSB sample.  Wondering if we should consider making a dedicated driver or keep as is for applications to clone.

jfischer-no, 2025-07-02:
> I am wondering if we should consider making a dedicated driver 

That is not possible because WebUSB does not describe a specific USB function/class. Thus, the device part cannot be reused in a generic way.

> or keep as is for applications to clone.

There are at least two samples where the header can be used as a template, https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/usb/webusb-next/src/webusb.h

closed 2025-07-02 by jfischer-no