zephyrproject-rtos/zephyr #12530: DTS: Changes done to support QSPI memory mapped flash breaks intel_s1000 build

rgundi, 2019-01-17:
The commit e57a596a0c7d2b40bd1baa46f72a04066bdbdaa4 breaks the build for intel_s1000. You can try building tests/boards/intel_s1000_crb and below is what you see.

Traceback (most recent call last):
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract_dts_includes.py", line 773, in <module>
    main()
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract_dts_includes.py", line 754, in main
    defs = generate_node_definitions()
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract_dts_includes.py", line 713, in generate_node_definitions
    flash.extract(node_address, 'zephyr,flash', 'FLASH')
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract/flash.py", line 161, in extract
    self._extract_flash(node_address, prop, def_label)
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract/flash.py", line 115, in _extract_flash
    default.extract(node_address, prop, None, def_label)
  File "/home/rgundi/RajZephyr/zephyr/scripts/dts/extract/default.py", line 36, in extract
    prop_values = reduced[node_address]['props'][prop]
KeyError: 'write-block-size'
CMake Error at /home/rgundi/RajZephyr/zephyr/cmake/dts.cmake:193 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/rgundi/RajZephyr/zephyr/cmake/app/boilerplate.cmake:296 (include)
  CMakeLists.txt:2 (include)

dcpleung, 2019-01-17:
It seems to be caused around line 81, where the `node_address` is being reassigned to the parent node. The `self._flash_node` still points to the actual flash node. However, when calling `default.extract()`, the `node_address` being passed points to the parent node, and `reduced[node_address]` no longer points to the same node as `self._flash_node`.

sathishkuttan, 2019-01-17:
> @sathishkuttan Do you know if the flash/SPI controller supports memory accessed to the flash on the Sue Creek SoC?

@galak Yes the SPI controller in Sue Creek SoC supports what is called XIP (eXecute In Place) mode where the serial flash can be accessed as any read-only memory mapped to the processor's address space

sathishkuttan, 2019-01-17:
Although hardware supports XIP, the SPI & flash drivers do not enable it. Hence the feature is not in S1000 port.

galak, 2019-01-17:
> Although hardware supports XIP, the SPI & flash drivers do not enable it. Hence the feature is not in S1000 port.

when you say the drivers don't enable it, are you saying its a pure software limitation?

sathishkuttan, 2019-01-17:
> when you say the drivers don't enable it, are you saying its a pure software limitation?

Yes. The driver implementation from @rgundi does not yet contain XIP support. At somepoint, he or someone may add code to use this mode.