zephyrproject-rtos/zephyr #12535: Bluetooth: suspend private address (RPA) rotating

Qbicz, 2019-01-17:
**Is your feature request related to a problem? Please describe.**
When Bluetooth privacy is enabled, address is changed every `BT_RPA_TIMEOUT` seconds.
However, for Microsoft Swift Pair feature it is needed to pause changing the address:

> If LE Privacy is supported, the peripheral should suspend rotating the Bluetooth LE Address during the Swift Pair session. The rotated address would be received as a new device request by Windows and would show two notifications for a single peripheral.

https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/bluetooth-swift-pair

**Describe the solution you'd like**
There could be API to stop/start the RPA change timer, which will be used by application to suspend the address change when Swift Pair advertising is used.

Vudentz, 2019-01-18:
I wonder if we should inhibit the rotation during the advertising if it is set as (limited) discoverable since this problem affects any stack that keeps a cache of devices while discovering (BlueZ keeps devices founds for 3 min.).

jhedberg, 2019-01-18:
> I wonder if we should inhibit the rotation during the advertising if it is set as (limited) discoverable since this problem affects any stack that keeps a cache of devices while discovering (BlueZ keeps devices found for 3 min.).

Doesn't sound like a bad idea. IIRC the discoverable mode is part of the advertising data (a "flags" data field), which bt_le_adv_start() doesn't look at currently.

Qbicz, 2019-01-22:
@jhedberg your simpler solution sounds good, what @Vudentz says is even simpler. I think it's about don't changing the RPA as long as device advertises (may be through resetting or blocking timer from expiring). The Kconfig option could be "Do not change the resolvable private address during advertising".

jhedberg, 2019-01-22:
> I think it's about don't changing the RPA as long as device advertises

@Qbicz 	I think that's making it too simple. There are devices that advertise all the time (e.g. beacons, but probably also some which are connectable). If the app developer enables privacy there should be a reasonable expectation that the device cannot be tracked over long periods of time. Suspending RPA rotation for discoverable advertising (where one of the discoverable flags is set in the advertising data) sounds like an interesting idea from @Vudentz . It'd be nice to hear some other opinions about it, e.g. from @sjanc @carlescufi @cvinayak and @holtmann

joerchan, 2019-03-25:
@jhedberg I think it would be better to force an RPA timeout instead of suspending it, that way we don't risk accidentally disabling privacy.
I think it is possible to do this with `bt_le_oob_get_local`

joerchan, 2019-06-12:
@Qbicz Does using `bt_le_oob_get_local` to generate a fresh address and reset the timer work for the swift pairing use-case?

Qbicz, 2019-06-12:
@joerchan this looks like a good idea, I will give it a try this week.

Qbicz, 2019-06-19:
Currently privacy does not work in our configuration because we use multiple identities. @MarekPieta will you be able to check it once problem with privacy/multiple identities is resolved?

joerchan, 2019-08-27:
@MarekPieta @Qbicz As far as I know the privacy issues have been fixed. Can you please check if the suggestion solves the problem?

MarekPieta, 2019-10-14:
I checked and I think we can close the issue. Thanks for all the help.

closed 2019-10-14 by joerchan