zephyrproject-rtos/zephyr #12537: potential over-use of k_spinlock

pabigot, 2019-01-17:
The `k_spinlock` API is intended to be used to preserve mutex by blocking interruptions.  In a uniprocessor context it's functionally equivalent to `irq_lock()`.  In a symmetric multiprocessor system its behavior is slightly different involving a secondary clear-and-set lock primitive that's specific to the spinlock instance.

`k_spinlock` was added concurrently with support for Tensilica (xtensa) processors.  It is primarily used in the scheduler and SoC-specific system timer modules, which are mostly uniprocessor architectures. A spinlock is also used in esp32 (xtensa) logging infrastructure.

There is no Kernel Services documentation on `k_spinlock`.  The Kernel Services documentation on preventing interruptions describes only `irq_lock()`, and specifies in particular that the lock is thread-specific, implicitly allowing the kernel code to context switch while the lock is held (with the possibility of releasing the lock if the new thread does not expect it to be held).

I have two concerns.

First, performing a context switch when a spinlock is held in an SMP system may result in a deadlock due to the secondary clear-and-set lock being held after the holding thread is swapped out and the global IRQ lock released.  If Zephyr developers see irq_lock and spinlock as roughly equivalent they may inappropriately use a spinlock in code where a context switch could occur while it's held.  This may be resolved by documentation.  See also #6521.

Second, using the spinlock API in architecture-specific code that can never run in an SMP situation obscures the implicit presence of `irq_lock()`, which is a pretty blunt instrument.  For example we've learned the Bluetooth subsystem on nRF51 requires its direct ISR to run [within 10 us to meet a 138 us hard deadline](https://github.com/zephyrproject-rtos/zephyr/issues/12342#issuecomment-452285493).

Somebody looking for reasons why Bluetooth is not working might reasonably search for `irq_lock`, and miss that the spinlock in `nrf_rtc_timer` blocks interrupts for 3.75 us in the ISR for a tickless kernel (longer if not tickless), and for up to 11.75 us observed in `z_clock_set_timeout()`.

pabigot, 2019-01-17:
@carlescufi @pizi-nordic FYI regarding the timer infrastructure potentially exceeding the nRF51 bluetooth hard deadline limits.  Not sure if that had been measured before.

carlescufi, 2019-01-17:
@pabigot thanks. No, we had not measured this in a while, last time we did we only had `irq_lock()`. This is obviously an issue, but it's difficult to solve if we want the kernel to be SMP-friendly in the future, so I think the clean solution is to have a dedicated documentation section called something like "Measuring interrupt latency" that describes the required steps to be performed to measure the longest interval with interrupts disabled.

@cvinayak 11.75us with interrupts disabled in `z_clock_set_timeout()` could be a real issue, correct?

pabigot, 2019-01-17:
Using `irq_lock()` makes no difference; they're equivalent in this case (confirmed by testing after switching the lock).  My point was that there was no benefit to switching to spinlock in this module, and doing so we make it harder to locate performance bottlenecks.

pabigot, 2019-01-17:
I've opened #12541 to track the Nordic-specific concern, as I don't want the more general issue I've raised to get lost in the noise.

andyross, 2019-01-17:
I'm not quite understanding the point here.  Spinlocks on uniprocessor systems resolve into an inline _arch_irq_lock().  It's exactly the same mechanism.  On SMP, they're required for proper semantics (and irq_lock() has a fallback implementation that preserves existing semantics but is quite a bit more complicated).  The expectation over time is that all code in the core kernel will be using this API exclusively, along with any drivers that might expect to be run in an SMP environment.

Is the bug here that you just want the nrf timer driver to use irq_lock()?

andyross, 2019-01-17:
OK, reading through the comments, it's all just a misconception.

Everywhere you see a k_spinlock_un/lock() call, the generated code is 100% equivalent to irq_un/lock().

pabigot, 2019-01-17:
@andyross This is not a bug (I've learned that lesson), nor is it necessarily something that should be assigned to you (if that's going to cause problems).  The nrf timer maintenance taking too long is a different issue should be addressed separately, probably by Nordic folks.

I opened this issue specifically for the points I made in the initial comment: that there is a difference of behavior for the two APIs in SMP environments that should be documented, and that casual use of the SMP-safe API in SoC-specific drivers that can never be SMP-enabled environment makes it difficult to find code that blocks all interrupts unless people know to also look for uses of `k_spin_lock`.

This concern is not bug, nor enhancement, nor RFC, nor feature request, nor (really) a question.  The failure to fit it into a fixed set of categories doesn't mean it's not important.

(Nor is it a misconception:
> Everywhere you see a k_spinlock_un/lock() call, the generated code is 100% equivalent to irq_un/lock().

I specifically said that equivalence held for uniprocessor.  I don't believe it does hold for SMP because of the spinlock-specific `atomic_t` lock, which would not be released by a context switch at least until #6521 is resolved.  Maybe at that point they'd be equivalent.)

nashif, 2020-02-14:
question answered.

closed 2020-02-14 by nashif