zephyrproject-rtos/zephyr #12544: commit 2fb616e broke I2C on Nucleo F401RE + IKS01A2 shield

avisconti, 2019-01-17:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

avisconti, 2019-01-17:
@erwango @galak
I do not have time to debug it.
Better if the author of the commit can take a look at it.
(@cybertale )

erwango, 2019-01-17:
@avisconti , what was nucleo reference?

avisconti, 2019-01-17:
> @avisconti , what was nucleo reference?

Nucleo F401RE

erwango, 2019-01-17:
`samples/shields/x_nucleo_iks01a1` is working on this same nucleo_f401re.
Do you test on this shield?

avisconti, 2019-01-17:
No, IKS01A2 
Look at #12398

avisconti, 2019-01-17:
> `samples/shields/x_nucleo_iks01a1` is working on this same nucleo_f401re.
> Can you test on this shield?

Ah, it's a request.
OK

erwango, 2019-01-17:
So, I had some tests and I finally have same status before and after incriminated SHA1.
I'll try to see if I can get hands on x_nucleo_iks01a2 shield, but in the mean time, it would be good you could have other tests on your side as well.

erwango, 2019-01-18:
@avisconti, I haven't found x_nucleo_iks01a2, so I've stripped out x_nucleo_iks01a2 sample to keep only hts221 code and I'm testing with x_nucleo_iks01a1 shield.
This is actually working even with 2fb616e commit.

Can you reproduce the error in similar conditions?
Or is the error specific to other sensor?

avisconti, 2019-01-18:
@erwango 
This morning I was about to push again the #11188 and before doing that I also rebase to the latest
commit of zephyr master. Well, I found that the 2fb616e is breaking also the microphone sample.
If I revert that commit everything is working again. So, I believe that there is something in that commit
that I need to investigate. I'm depressed..... but hopefully it will not tke that long.

explora26, 2019-01-21:
Hi, the 2fb616efbef663a5912673507031e3cf102ec828 also breaking the hello world example on stm32_min_dev board, there is nothing output from UART1.

erwango, 2019-01-21:
@explora26 , I'm working on a fix. Can you test with https://github.com/erwango/zephyr/commits/fix_gpio and see if solving your issue?

explora26, 2019-01-21:
@erwango , I just test it, but the issue still not fixed.

avisconti, 2019-01-21:
@erwango 
In my case it is fixed, at least the microphone sample. But I'm confident also for the other setup.

@explora26 
What STM32 architecture you are based on?

erwango, 2019-01-21:
@avisconti , @explora26 , I've pushed a clean up commit in https://github.com/erwango/zephyr/commits/fix_gpio , if you have some time to confirm it still works would be nice.
I'll do full non regression and then push a PR later today

explora26, 2019-01-21:
@avisconti , It's stm32f103c8t6.

@erwango , I tested it again, but unfortunately it's still not work.

erwango, 2019-01-21:
@explora26 , if you have access to gdb, can you do the following before and after the change?
```
(gdb)x /4x 0x40010800
0x40010800:	0x44444944	0x88844444	0x0000dfdc	0x0000a000
```

pabigot, 2019-01-21:
@erwango +1 for getting your `fix_pgio_refactor` branch merged.  It resolves the inability to see any I2C devices on `nucleo_l476rg` that I lost significant time trying to understand when testing #12567.

explora26, 2019-01-22:
@erwango 
Here is the registers dump of https://github.com/erwango/zephyr/commits/fix_gpio
```
(gdb) x /4x 0x40010800
0x40010800:	0x44444444	0x88844444	0x0000dffc	0x0000a000
```

erwango, 2019-01-22:
@explora26, txs.
Is that dump done at boot or following break in Idle (with Crtl-C)?
Can you do the same procedure before the commit that introduced the issue?

PR #12623 (2019-01-23): drivers/gpio: stm32: Fix gpio_stm32_configure
Behavior of function gpio_stm32_configure has been modified
during driver factorization. Various gpio settings (speed, mode, ..)
are applied conditionally while they used to be applied in sequence,
unconditionally before this change. As a consequence some
combinations of configurations are no more applied (like speed for
alternate mode). This of course has impact in some use cases.
Rework functions in order to apply settings unconditionally. Take
advantage of the change to reduce code size.

This change impacts all SoCs except F1 series.

Fixes #12 [... cut]
commit 94bf757d (2019-01-23): drivers/gpio: stm32: Fix gpio_stm32_configure

Behavior of function gpio_stm32_configure has been modified
during driver factorization. Various gpio settings (speed, mode, ..)
are applied conditionally while they used to be applied in sequence,
unconditionally before this change. As a consequence some
combinations of configurations are no more applied (like speed for
alternate mode). This of course has impact in some use cases.
Rework functions in order to apply settings unconditionally. Take
advantage of the change to reduce code size.

This change impacts all SoCs except F1 series.

Fixes #1 [... cut]
closed 2019-01-23 by galak