zephyrproject-rtos/zephyr #12548: ISR sometimes run with the MPU disabled: breaks __nocache

aurel32, 2019-01-17:
**Describe the bug**

On ARM platforms with and `CONFIG_CPU_HAS_ARM_MPU=y` and `CONFIG_MPU_STACK_GUARD=y`, the ISR might sometimes (but rarely) run with the MPU disabled.

This happens because the MPU is briefly disabled and then re-enabled in `configure_mpu_stack_guard`, which is called during context switch: https://github.com/zephyrproject-rtos/zephyr/blob/962f900c0b73878545e47c6635fe27bf4d8102f7/arch/arm/core/cortex_m/mpu/arm_core_mpu.c#L38-L41

If an interrupt happens during this small amount of time, it runs with the MPU disabled. The MPU is used by the __nocache feature to disable cache on a memory region. When the MPU is disabled, the cache is enabled again. Access to this regions therefore sometimes returns stale data from the previous time the cache was enabled.

The workaround it to set `CONFIG_MPU_STACK_GUARD=n` and `CONFIG_USERSPACE=n`. This avoid the MPU to be disabled and re-enabled during context switch.

**To Reproduce**
This was observed when working on PR #11888. The problem happens when doing multi-Mb transfer on the ethernet interface, therefore causing a lot of interrupts. I used this debug code in the interrupt, I guess it can also be reproduced with other interrupts:

```
if ((MPU->CTRL & MPU_CTRL_ENABLE_Msk) == 0) {
        printk("MPU is disabled\n");
}
```

**Expected behavior**
The MPU is always enabled when running ISR code.

**Impact**
This breaks __nocache support, which in turns prevents to enable cache on SAM E70 when the ethernet driver is enabled (PR #11888).

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Binutils/GCC from Debian sid.
 - Commit SHA: 689ad43b6207e0238746a2e7513849d355037686

aurel32, 2019-01-17:
@andrewboie @andyross @ioannisg @angansari You might be interested by this issue.

andrewboie, 2019-01-17:
We agreed to get rid of the disable/enable calls.

ioannisg, 2019-01-23:
> 
> 
> We agreed to get rid of the disable/enable calls.

This will be handled in the ARM MPU re-work (for 1.14)

aurel32, 2019-01-23:
> > We agreed to get rid of the disable/enable calls.
> 
> This will be handled in the ARM MPU re-work (for 1.14)

This currently blocks PR #11888. Can we agree to also have that one merged for 1.14?

ioannisg, 2019-01-23:
> 
> 
> > > We agreed to get rid of the disable/enable calls.
> > 
> > 
> > This will be handled in the ARM MPU re-work (for 1.14)
> 
> This currently blocks PR #11888. Can we agree to also have that one merged for 1.14?

That's ok with me