zephyrproject-rtos/zephyr #12559: tests/kernel/mem_pool/mem_pool_threadsafe/kernel.memory_pool fails sporadically

nashif, 2019-01-18:
**Describe the bug**
test fails sporadically on both qemu_x86 and mps2_an385

**To Reproduce**
sanitycheck -T tests/kernel/mem_pool/mem_pool_threadsafe

```
Booting from ROM..***** Booting Zephyr OS zephyr-v1.13.0-3307-g12f984995f *****
Running test suite mpool_threadsafe
===================================================================
starting test - test_mpool_threadsafe
ASSERTION FAIL [block_fits(p, block, lsz)] @ /home/nashif/github/zephyr/lib/mempool/mempool.c:162

```

on mps2_an385 (qemu):
```
***** Booting Zephyr OS zephyr-v1.13.0-3307-g12f984995f *****
Running test suite mpool_threadsafe
===================================================================
starting test - test_mpool_threadsafe
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x300005b0
***** Hardware exception *****
Current thread ID = 0x2000009c
Faulting instruction address = 0x4752
Fatal fault in ISR! Spinning...
```

andrewboie, 2019-02-04:
Seen this also on qemu_nios2

andyross, 2019-02-26:
Partially addressed by #13770.  Seems like all emulation platforms but ARM are fine now.  There, though, the test shows a hilarious inversion of our normal pattern: it works well enough in a big sanitycheck, but if I run it in isolation I get pretty reproducible deadlocks.  Still looking.

andyross, 2019-02-27:
Seems to be down to the swap-is-async priority bug detailed in #12553.  Goes away with the trial patch in that bug.

andyross, 2019-03-07:
See extensive notes in #12553.  This test, on mps2_an385, is the only spurious that I can't make go away in volume testing with careful control over machine load and host kernel tick rate.  It's likely a real bug somewhere.

andyross, 2019-03-14:
Fix up in #14539 has crossed 5500 separate test cases now (790+ runs across the 8 default boards/variants) with no failures.  Think this one is fixed.

andyross, 2019-03-15:
Fixed merged, but forgot "Fixes ..." in commit.  Oops.

closed 2019-03-15 by andyross