zephyrproject-rtos/zephyr #12560: Using TCP w/ wired NIC results in mismanagement of buffers due to ACK accounting error

agross-linaro, 2019-01-18:
This bug manifested itself originally as MQTT connection errors when I was trying to connect to a broker.  Sometimes the connection would work, sometimes it would fail.  I turned on network debugging and starting seeing buffer low messages.  After trying to debug this and being stymied by not knowing what really is going out on the wire, I bought a managed switch and set up port mirroring.  That is when I realized what was happening.  A week later, after peeling apart the network stack I found the issue.

The TCP protocol stack we are using will hold on to packets in their entirety (net_pkt and associated net_bufs) until we receive an ACK from the other host.  If this times out, we send retries and basically replay the packet.  In the L2 subsystem, when we send packets, we pre-pend 14 bytes of header to the packet.  This has rather dire consequences when the ACK comes in or if you need to retransmit.

When the ACK comes in, we look in our list of sent packets and we index into the net bufs to find the sequence number.  Unfortunately, due to the above prepending of the 14 bytes, this is no longer where we think it is.  We then pull out a number which is part of the MAC address and use that for our comparison.  If you are 'lucky' and your ack is higher than this number, you will automatically clear the sent packets.  If you are 'unluckly', you will not retire any packets.  Eventually, you will bleed out your buffers and the system will crash or end up malfunctioning in a variety of ways.

If you are lucky enough to have the retry timer fire off, you will then retry and prepend ANOTHER L2 header and send out a malformed packet.

Frankly, I don't know how this EVER worked.  This has caused me 2 weeks of pain, thinking that the MQTT work (the real work I want to do) was the issue.  I don't know how this ever passed testing.  I did not spend any time doing any type of historical analysis of this code.  I don't really care to as I already have spent an inordinate amount of time.

You can reproduce this on any system with a wired NIC.  It may reproduce on others, who knows.  I used a FRDM K64F.  You can probably take the echo client and echo server and run them in TCP and roll the dice.

agross-linaro, 2019-01-18:
Referencing MQTT app where I encountered this issue.
https://github.com/zephyrproject-rtos/zephyr/pull/10500

tbursztyka, 2019-01-18:
>  I don't know how this ever passed testing.

Because there is nothing that tests that.

Hopefully the fix is easy.

pfalcon, 2019-01-18:
@agross-linaro :

> I don't know how this EVER

Welcome to the club ;-). The change of prepending Eth headers was merged relatively recently, literally a week or two before Xmas holidays, may explain why other people didn't hit it so hard. I for one recently tested HTTP client and server samples and didn't notice anything unusual - they work subideal as they usually did, eventually often converging to stuck stack. Which is in sharp contrast to @jukkar's reports, for whom everything usually works ;-).

pfalcon, 2019-01-18:
FWIW, I can confirm some inadequacy with frdm_k64f and dumb_http_server. But from quick try for me it's not "it's broken and doesn't work", but "it's even lousier than it's usually". The usual picture is described in https://github.com/zephyrproject-rtos/zephyr/pull/11530 (open for 9 months total). Now I get:

~~~
Time taken for tests:   67.393 seconds
Complete requests:      43

...

Percentage of the requests served within a certain time (ms)
  50%   1664
  66%   2112
  75%   2464
  80%   2560
  90%   2912
  95%   3040
  98%   3814
  99%   3814
 100%   3814 (longest request)
~~~

I.e. whereas usually response slowdown happens after ~50 of requests, with current master (eff81c498b55925d2755a2d2eaef601d0461e2ec), it happens after ~20.

pfalcon, 2019-01-18:
Also tried big_http_download sample on frdm_k64f. It also was kinda lurking, definitely slower than usual. Seems to finally stuck at:

~~~
1670170 bytes
~~~

All in all, that's my trying to figure out how it's possible that nobody noticed the issue reported by @agross-linaro. Well, it's possible that other stuff mostly works, and as there're already known issues, the effect of new issues isn't immediately recognizable.

That's not to say that we don't need more tests. The opposite. See e.g. https://github.com/zephyrproject-rtos/zephyr/issues/3559, submitted 1.5+ years ago, never receiving a comment from anyone.

carlescufi, 2019-01-22:
@tbursztyka do you expect to be able to submit a fix for this issue?

agross-linaro, 2019-01-24:
@carlescufi https://github.com/zephyrproject-rtos/zephyr/pull/12563

agross-linaro, 2019-01-24:
However, it seems like there are still additional issues w/ that patch.

galak, 2019-02-05:
@jukkar @tbursztyka any updates on this?

jukkar, 2019-02-06:
The issue is being worked on in #12563 right now.

PR #12563 (2019-02-08): net/ethernet: Remove inserted L2 header buffer
The packet can be referenced somewhere else and letting the newly added
L2 header will generate corrupt packet. If the same packet is being
resent, ethernet will add again its L2 header. Thus the need to remove
such L2 header every time a packet has been sent, successfully or not.

Fixes #12560
commit 499ffd65 (2019-02-08): net/ethernet: Remove inserted L2 header buffer

The packet can be referenced somewhere else and letting the newly added
L2 header will generate corrupt packet. If the same packet is being
resent, ethernet will add again its L2 header. Thus the need to remove
such L2 header every time a packet has been sent, successfully or not.

Fixes #12560
closed 2019-02-08 by jukkar