zephyrproject-rtos/zephyr #12562: The return value of net_buf_simple_pull is unreasonable in subsys/net/buf.c

xiaoliang314, 2019-01-18:
**Describe the bug**
The return value of net_buf_simple_pull is unreasonable. It should pull data from the buffer and return it. However, it discards this data and returns a new address.
```
void *net_buf_simple_pull(struct net_buf_simple *buf, size_t len)
{
	NET_BUF_SIMPLE_DBG("buf %p len %zu", buf, len);

	NET_BUF_SIMPLE_ASSERT(buf->len >= len);

	buf->len -= len;
	return buf->data += len;
}
```

jhedberg, 2019-01-18:
The net_buf APIs have originally been designed to be semantically similar to the Linux kernel skbuff APIs. In this case the behaviour is simply reflecting that of skb_pull():
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/linux/skbuff.h#n2153

I'm not sure it's worth doing the change since it may break existing code outside of the tree, and it will also confuse anyone who is familiar with Linux networking development.

@jukkar @tbursztyka FYI

xiaoliang314, 2019-01-18:
Thank you for your answer, I think it should be modified. These are my reasons:
1. In the Bluetooth protocol stack, net_buf is the main data structure that provides data to users, which means that net_buf APIs will be used by more ordinary users. It may cause confusion for more people.
2. Many places use this API in this way
···
label_uuid = buf->data;
net_buf_simple_pull(buf, 16);
···

xiaoliang314, 2019-01-18:
Thank you for your answer, I think it should be modified. These are my reasons:
1. In the Bluetooth protocol stack, net_buf is the main data structure that provides data to users, which means that net_buf APIs will be used by more ordinary users. It may cause confusion for more people.
2. Many places use this API in this way
```
label_uuid = buf->data;
net_buf_simple_pull(buf, 16);
```

jukkar, 2019-01-22:
I think we should not touch the existing function as it is already in use. But I suppose if needed we could introduce a new function that could do it like that.