zephyrproject-rtos/zephyr #12568: ST: Is it impossible to flash using ST-Link/V2?

xoduddk123, 2019-01-18:
Hi.

I am a starter of Zephyr.
I am trying to run the `STM32F415RG` with Zephyr.

My target board is [MINI-M4 for STM32](https://www.mikroe.com/mini-stm32f4).
The `MINI-M4` does not include ST-Link.
I have `ST-Link/V2`.
So I connected the `ST-Link/V2` and the `MINI-M4` as shown below.
[image]

When I connect `ST-Link/V2`, `/dev/ttyACM0` is not generated.
When [STM32F4DISCOVERY](https://www.st.com/en/evaluation-tools/stm32f4discovery.html) is connected, `/dev/ttyACM0` is created.
Is `/dev/ttyACM0` required for make flash?

I could not find a place to sell `ST-Link/V2-1`.
I saw the following [document](https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/tools/stlink-v21.rst).
> Use the CMake flash target with the argument STLINK_FW=jlink to build your Zephyr application.

Is it possible to `make flash` on [MINI-M4 for STM32](https://www.mikroe.com/mini-stm32f4) using `J-Link`?

Below is my build command and `make flash` log.
```
mkdir sample/hello_world/build
cd sample/hello_world/build
cmake -DBOARD=stm32f4_disco ..
make -j8
make flash

root:build# make flash
[  1%] Built target kobj_types_h_target
[  2%] Built target syscall_macros_h_target
[  5%] Built target syscall_list_h_target
[  6%] Built target driver_validation_h_target
[  8%] Built target offsets
[  9%] Built target offsets_h
[ 28%] Built target kernel
[ 29%] Built target app
[ 30%] Built target linker_script
[ 55%] Built target zephyr
[ 68%] Built target arch__arm__core
[ 74%] Built target arch__arm__core__cortex_m
[ 77%] Built target arch__arm__core__cortex_m__mpu
[ 87%] Built target lib__libc__minimal
[ 88%] Built target boards__arm__stm32f4_disco
[ 89%] Built target drivers__gpio
[ 91%] Built target drivers__pwm
[ 93%] Built target drivers__serial
[ 95%] Built target zephyr_prebuilt
[ 95%] Built target linker_pass_final_script
[ 98%] Built target kernel_elf
[100%] Flashing stm32f4_disco
Using runner: openocd
Open On-Chip Debugger 0.10.0+dev-gea2753a7-dirty (2018-11-02-23:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v32 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.217357
Error: init mode failed (unable to connect to the target)
in procedure 'init' 
in procedure 'ocd_bouncer'

ERROR: command exited with status 1: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /work/zephyr/boards/arm/stm32f4_disco/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /work/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset halt' -c 'verify_image /work/zephyr/samples/hello_world/build/zephyr/zephyr.elf' -c 'reset run' -c shutdown
run as "west -v ... flash ..." for a stack trace
[100%] Built target flash
root:build# 
```

I'm sorry I do not have enough English.
Thank you.

erwango, 2019-01-18:
@xoduddk123 , STM32F415RG SoC is not supported yet, you should first enable it's support in Zephyr.
Then, you should port your board since it is not supported in zephyr.
Regarding ST-Link/V2, I think you should be able to use it. One thing to check is stm32f415 support in openocd.

xoduddk123, 2019-01-18:
@erwango 
I know that `STM32F415RG` and` MINI-M4` are not supported in Zephyr.
My goal is to support `STM32F415RG` in Zephyr.
To me this is an interesting way to study Zephyr.
I was looking at #9333 and trying `STM32F415RG`.

I do not know what `openocd` is yet.
However, I found a commit of [cfg: add STM32F4x and STM3241G-EVAL config files](https://github.com/zephyrproject-rtos/openocd/commit/e9976aa3a879ac45dc79a96f6a87cc9fcdf7aaca#diff-ad88f0c757699ff75ef54138f3d2ebbd).
Should it be added to `openocd` in the form below?

#### tcl/board/mini-m4_for_stm32.cfg
``` cfg
source [find interface/stlink-v2.cfg]

# increase working area to 192KB
set WORKAREASIZE 0x30000

# chip name
set CHIPNAME STM32F415RG

source [find target/stm32f4x_stlink.cfg]
```

If it is not a problem of `ST-Link/V2`, can I make a new issue called `STM32F415RG support`?

xoduddk123, 2019-01-19:
@erwango 
My cable connection was missing `RST` and `GND`.
I connected ST-Link/V2 and MINI-M4 again as shown in the table below.
## ST-Link/V2 - MINI-M4
| ST-Link/V2(Pin no.)   | MINI-M4(Pin no.)    | 
| --------------------- | ------------------- |
| TCK_SWCLK(9)          | PA14(36)            |
| TMS_SWDIO(7)          | PA13(35)            |
| TDI(5)                | PA15(30)            |
| TDO_SWO(13)           | PB3(29)             |
| GND(8)                | GND(31)             |
| NRST(15)              | MCLR(1)             |

[image]

I downloaded and installed `openocd`.
I created the following cfg file.
``` cfg
source [find interface/stlink-v2.cfg]

# increase working area to 192KB
set WORKAREASIZE 0x30000

# chip name
set CHIPNAME STM32F415RG

source [find target/stm32f4x_stlink.cfg]
```

Then I ran `openocd` with the following command:
``` bash
root:~# 
root:~# 
root:~# /usr/local/bin/openocd -f /usr/local/share/openocd/scripts/board/mini-m4.cfg 
Open On-Chip Debugger 0.10.0+dev-00656-gd140fb27 (2019-01-19-13:58)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
WARNING: target/stm32f4x_stlink.cfg is deprecated, please switch to target/stm32f4x.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J32S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.211556
Info : STM32F415RG.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
```

It seems that `openocd` supports STM32F415.