zephyrproject-rtos/zephyr #12570: Zephyr codebase incorrectly uses #ifdef for boolean config values

pkral78, 2019-01-18:
This is widespread around whole Zephyr codebase. 

Instead of proper usage of `#if` directive, `#ifdef` is used.

Although this has no visible impact on functionality, because autoconf.h contains only those symbols that have true (1) value (thus are defined) this is sematically incorrect.

SebastianBoe, 2019-01-18:
Can you elaborate on what problems this causes, and how this could possibly be rectified / enforced ?

pkral78, 2019-01-18:
Hmm after further thinking about this and realization that only possible harm is in form of generating boolean symbol with zero values (`#define CONFIG_BOOL_XXX 0`) Which probably would never happened I'm going closing this as offtopic.

closed 2019-01-18 by pkral78