zephyrproject-rtos/zephyr #12571: No coverage reports are being generated

nashif, 2019-01-18:
**Describe the bug**

Running sanitycheck with --coverage fails to generate reports

**To Reproduce**

`sanitycheck --coverage --coverage-platform qemu_x86 -p qemu_x86 -T tests/kernel/`

**Expected behavior**
Full report available

**Impact**

Recently added coverage support becomes unusable

**Screenshots or console output**

```
.dir/arch/x86/core/offsets/offsets.c.gcno:version '602*', prefer 'A82R'
/home/nashif/Work/github/zephyr/sanity-out/qemu_x86/tests/kernel/msgq/msgq_api/kernel.message_queue/zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.gcno:no functions found
/home/nashif/Work/github/zephyr/sanity-out/qemu_x86/tests/kernel/msgq/msgq_api/kernel.message_queue/zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.gcda:version '602*', prefer version 'A82R'
geninfo: WARNING: gcov did not create any files for /home/nashif/Work/github/zephyr/sanity-out/qemu_x86/tests/kernel/msgq/msgq_api/kernel.message_queue/zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.gcda!
lcov: ERROR: no valid records found in tracefile /home/nashif/Work/github/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/nashif/Work/github/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/nashif/Work/github/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/nashif/Work/github/zephyr/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/nashif/Work/github/zephyr/sanity-out/coverage.info
73 of 76 tests passed with 0 warnings in 102 seconds

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK and host toolchain gcc 8.2.1
 - Commit zephyr-v1.13.0-3330-geff81c498b

AdithyaBaglody, 2019-01-21:
@nashif I am not able to see this issue at 2caecb54a37dca72fdd20aa45ac95e4eeeaf0adb (the latest master). 
Can you check if it is failing still?

carlescufi, 2019-01-22:
@nashif can you check if this is still an issue?

nashif, 2019-01-22:
> Can you check if it is failing still?

Please make sure you are using a recent host toolchain, gcc 8.2 or something.

nashif, 2019-01-23:
also note that when building with the beta SDK 0.10.0-beta3 we get this:

```
/home/buildslave/src/github.com/zephyrproject-rtos/sdk-ng/build/build_i586/.build/i586-zephyr-elf/src/gcc/libgcc/libgcov-driver.c:903: multiple definition of `__gcov_init'; libzephyr.a(coverage.c.obj):/home/nashif/Work/github/zephyr/tests/coverage/coverage.c:35: first defined here
```

nashif, 2019-01-23:
could be related?

gcc-mirror/gcc@a82502d
gcc-mirror/gcc@3cf7fdd

AdithyaBaglody, 2019-01-23:
@nashif The beta SDK is incompatible with the current zephyr. Provide all the information and the patches that are needed to do a simple build.

nashif, 2019-01-23:
> @nashif The beta SDK is incompatible with the current zephyr. Provide all the information and the patches that are needed to do a simple build.

the beta SDK works with the xtools variant

`export ZEPHYR_TOOLCHAIN_VARIANT=xtools`

nashif, 2019-01-23:
the issue with the new SDK is another issue, the original issue is reproducible with the current SDK and gcov from recent toolchains.

AdithyaBaglody, 2019-01-23:
> the original issue is reproducible with the current SDK

Not possible. i tested on multiple machines. with the current SDK

>the beta SDK works with the xtools variant

I dont have this "Xtools"

nashif, 2019-01-23:
> Not possible. i tested on multiple machines. with the current SDK

yes possible. This is an issue I reported already last year and I demonstrated to you how it did not work on my Fedora 29. Did you try on Fedora 29? What is your host distro and version? What is the toolchain version of your host?

> I dont have this "Xtools"

xtools = the beta SDK you download from https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.10.0-beta4. instead of zephyr as variant you set xtools and point XTOOLS_TOOLCHAIN_PATH to where the sdk was installed.

AdithyaBaglody, 2019-01-24:
Few things to note here. 
1. Sanitycheck script is doing a different build. I am not sure what it is. But when I run manually with coverage enabled. the build passes.
2. GCC counters for gcov has changed and this will be made into a PR.
3. sanitycheck should always use the gcov binary available in the SDK. It will cause incompatibility when we use different version of gcc and gcov.
(FYI I tested this on fedora 29 with 
` gcovr -r . --html -o gcov_report/coverage.html --html-details -vvvvv --gcov-executable $<path to SDK GCOV>`
    This way we can keep the host tool chains out. I assumed this is how the sanity was doing it till now. Please correct it if not.
4. The build flags of i586 gcc is different from that of arm and that of the older i586. (@nashif can you please check that they are as expected.) 
       The problem is that the build is not including the gcov hooks like it does on ARM. With the point no2 change, i am able to get reports perfectly on ARM. On x86 the gcov hooks are missing. Please check if the build of the i586 gcc is correct.

nashif, 2019-01-24:
> 1. Sanitycheck script is doing a different build. I am not sure what it is. But when I run manually with coverage enabled. the build passes.

ok, this is with the new SDK, right? I think this is the cause:

```
        if options.enable_coverage:
            args += ["EXTRA_LDFLAGS=--coverage"]
```

AdithyaBaglody, 2019-01-25:
>Need more details on this one, can you please elaborate. 

@nashif 
I am not able to pin point exactly what has to be added into the build of i586. But the same code works on arm mps2_an385. 
I am guessing here, but we might have to enable-multilib with-multilib-list 32,64,x32 libraries. and enable-thread as posix. 

The problem is quite strange, the hooks of gcov are not getting placed in the code. But gcno files are getting generated. therefore I was guessing the build of i586 gcc might have a problem.

> GCC counters for gcov has changed and this will be made into a PR.

I will create a PR now and you can have a test with the updated sanitycheck.

nashif, 2019-01-28:
> @nashif
> I am not able to pin point exactly what has to be added into the build of i586. But the same code works on arm mps2_an385.
> I am guessing here, but we might have to enable-multilib [...]

Need to continue looking into this, not able to generate coverage for x86 is no-go. I will try a few things on the toolchain side, but please try to rootcause the issue on your side as well.

AdithyaBaglody, 2019-01-29:
>Need to continue looking into this, not able to generate coverage for x86 is no-go. I will try a few things on the toolchain side, but please try to rootcause the issue on your side as well.

Any breakthrough on the i586 gcc build??

nashif, 2019-01-30:
closing, only remaining issue is with new toolchain and x86, can be addressed with host toolchain for now.

closed 2019-01-30 by nashif