zephyrproject-rtos/zephyr #12589: Several nRF based boards enable both I2C & SPI by default in dts at same MMIO address

galak, 2019-01-18:
The following warnings show cases for various nRF based boards in which 2 controllers at the same MMIO address are both enabled by default.

This warning is generated with the DTC version that is in sdk 0.10-beta4:

96b_nitrogen.dts.pre.tmp:101.22-110.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40003000: duplicate unit-address (also used in node /soc/spi@40003000)
nrf51_pca10028.dts.pre.tmp:107.22-116.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40004000: duplicate unit-address (also used in node /soc/spi@40004000)
nrf52840_blip.dts.pre.tmp:134.22-143.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40004000: duplicate unit-address (also used in node /soc/spi@40004000)
nrf52840_pca10056.dts.pre.tmp:134.22-143.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40004000: duplicate unit-address (also used in node /soc/spi@40004000)
nrf52840_pca10059.dts.pre.tmp:134.22-143.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40004000: duplicate unit-address (also used in node /soc/spi@40004000)
nrf52_pca10040.dts.pre.tmp:111.22-120.5: Warning (unique_unit_address_if_enabled): /soc/i2c@40004000: duplicate unit-address (also used in node /soc/spi@40004000)

galak, 2019-01-18:
Here's the list of boards impacted:

96b_nitrogen nrf51_pca10028 nrf52840_blip nrf52840_pca10056 nrf52840_pca10059 nrf52_pca10040

pabigot, 2019-01-19:
The defaults on those boards should probably be updated to eliminate the conflict.

But at a higher level it's actually OK to have drivers for TWIx and SPIx instances that share a peripheral block available in the kernel, as long as the peripherals are not both enabled simultaneously.  At a cursory glance it appears TWI instances are enabled only for the duration of the bus transfer: once that completes, they're disabled.  SPI instances appear to be enabled on initialization and never disabled, which should be reconsidered if it has power consumption impacts.

Applications on Nordic hardware may legitimately be designed to switch between the TWI and SPI personalities of a given peripheral block based on transient needs.  In that case, the warning from the new dts would be appropriate but ignored.  (All that to say: however this is resolved, please don't remove from the application design space that option of overloading peripheral block roles.)

carlescufi, 2019-01-21:
@anangl can you please weigh in here?

carlescufi, 2019-03-05:
@anangl could you please check this one?

anangl, 2019-03-05:
> @anangl could you please check this one?

OK, I will take a (closer) look.

anangl, 2019-03-06:
In #14097 I altered the dts files for the boards mentioned above. This should prevent the warnings. But this is more a workaround then a proper solution for this problem. As @pabigot already mentioned, it may be desirable to enable both peripherals occupying the same block, let's say I2C_0 and SPI_0, and use them both, but of course not at the same time. We should provide such possibility.
A similar problem is with CLOCK and POWER peripherals. They also occupy the same block. And for example for nRF9160, we need to enable DT nodes for both of them, to get their base addresses mapped accordingly in secure or non-secure address space.
Would it be suitable to use something like this:
```
peipheral_block@5000 {
	reg = <0x5000 0x1000>;
	interrupts = <5 1>;

	clock {
		compatible = "nordic,nrf-clock";
		status = "ok";
		label = "CLOCK";
	};

	power {
		compatible = "nordic,nrf-power";
		status = "ok";
		label = "POWER";
	};
};
```
and get base addresses for them from the parent node? I guess some addition in the extracting script would be needed to accomplish this. And then we'll need also to provide some smart way of sharing the interrupt handler between the two drivers.
@galak Is there a "standard" way of dealing with such situations?

closed 2019-03-06 by galak

pabigot, 2019-03-06:
I see existing practice in Linux, e.g. [for nand devices with partitions](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm/boot/dts/am335x-evm.dts#n534), so this approach makes sense, disregarding the complexity of passing the information from the tree through the tooling to the kernel.

The parent name should be something like `clock_power@5000` (matching nrfx/mdk's identification of the corresponding block's IRQ) since the node names should be informative and must be unique.  Also then `uarte0_spim0_spis0_twim0_twis0@8000` for the largest conflict set I've seen so far.

This solution would also be nice because then all the potential personalities are co-located in the dts and a comment can be added to highlight the constraint on simultaneous operation.

PR #14097 (2019-03-06): boards: nrf: Prevent DTC warnings "duplicate unit-address"
**boards: 96b_nitrogen: Enable by default SPI_1 instead of SPI_0**

SPI_0 cannot be used simultaneously with I2C_0, since in nRF52832 SoC
these instances share certain hardware resources.

**boards: nrf: Correct I2C and SPI instances enabled by default in dts**

Do not enable by default both I2C and SPI nodes for the peripherals
with the same instance number, since they use the same MMIO base
address and DTC will issue a warning in such situation.
This patch corrects dts files for the following boards:
- nrf51_pca10028
- nrf52_pca10040 [... cut]