zephyrproject-rtos/zephyr #12594: stm32_min_dev board no console output

explora26, 2019-01-19:
I'm trying to test hello world example with stm32_min_dev board, but the UART didn't output anything.

I found that the last workable commit  SHA is 9612f9d840e7cb538817faac57ee461dcc9c6d8a, maybe this issue cause by 2fb616efbef663a5912673507031e3cf102ec828

erwango, 2019-01-21:
@explora26 , if you have access to gdb, can you do the following before and after the change?
```
(gdb)x /4x 0x40010800
0x40010800:	0x44444944	0x88844444	0x0000dfdc	0x0000a000
```

erwango, 2019-01-21:
^^@cybertale

explora26, 2019-01-22:
Here is the registers dump:
Before revert 2fb616efbef663a5912673507031e3cf102ec828 :
```
(gdb)x /4x 0x40010800
0x40010800:	0x44444444	0x88844444	0x0000dfff	0x0000a000
```
After revet 2fb616efbef663a5912673507031e3cf102ec828 :
```
(gdb)x /4x 0x40010800
0x40010800:	0x44444444	0x88844494	0x0000dfff	0x0000a000
```
Looks like the port configuration register incorrect.

erwango, 2019-01-22:
@explora26, great txs! This piece of code was indeed my suspect number 1, but I couldn't spot the issue.
Would you push a PR to fix it?

explora26, 2019-01-22:
@erwango , the pull request have been created, , thanks.