zephyrproject-rtos/zephyr #12601: MbedTLS-generic: Add setting for MBEDTLS_AES_ROM_TABLES config

mike-scott, 2019-01-20:
**Is your feature request related to a problem? Please describe.**
Many devices are very resource constrained.  By enabling:
MBEDTLS_AES_ROM_TABLES in the mbedtls config, MbedTLS uses less RAM and more flash.

For comparison here is reel_board before and after this MbedTLS setting is enabled:
[before]
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      269900 B         1 MB     25.74%
            SRAM:       71428 B       256 KB     27.25%
        IDT_LIST:         136 B         2 KB      6.64%
```
[after]
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      278108 B         1 MB     26.52%
            SRAM:       62692 B       256 KB     23.92%
        IDT_LIST:         136 B         2 KB      6.64%
```

**Describe the solution you'd like**
Would be great if we could have a Kconfig option for enabling this in:
ext/lib/crypto/mbedtls/Kconfig.tls-generic
Which would then be #ifdef'd in:
ext/lib/crypto/mbedtls/configs/config-tls-generic.h

**Describe alternatives you've considered**
It's possible that we could enable this config all the time (w/o a Kconfig) if the security folks weigh in on the downsides?

mike-scott, 2019-01-20:
I forgot to mention that I'm enabling this change for small-ish devices using DTLS encryption with PSK-only.  Combined with the following config changes, it makes a HUGE difference in RAM usage:
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED=n
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
(The above removes the cert parsing RAM structures)

pfalcon, 2019-01-20:
+1

Also +1 for making this default. May guess it still makes sense to make it configurable. Would need to check mbedTLS impl, but the usual choice is less RO data in the image, but startup initialization overhead. This makes sense e.g. for executables on conventional OSes (e.g. consider Linux executable can be made 90K instead of 100K), but usually not relevant for now-conventional XIP ROM MCUs. But there're still "non-XIP ROM image, large RAM" style MCUs, e.g. 1MB RAM, 256KB ROM. Then we're effectively constrained by ROM re: how much we can put on it even though RAM is huge, so option to save on ROM image size may be helpful.

@rlubos maintains new mbedTLS config infra, so hopefully he can comment on this.

rlubos, 2019-01-21:
Agreed, it makes sense to make this option configurable. Whether it is enabled by default is not a big deal IMHO, but as we do target RAM-constrained devices mostly, I guess we can enable it.

PR #12619 (2019-01-22): ext: lib: crypto: Extend generic mbedTLS config
Extend generic mbedTLS configuration file with MBEDTLS_AES_ROM_TABLES
option. This allows to save some RAM (~8kB) in favour of ROM and
performance.

Resolves #12601.
closed 2019-01-22 by nashif