zephyrproject-rtos/zephyr #12602: STM32F415RG Support

xoduddk123, 2019-01-20:
Hi.

I have a [MINI-M4 for STM32](https://www.mikroe.com/mini-stm32f4) board.
This board uses [STM32F415RG](https://www.st.com/en/microcontrollers/stm32f415rg.html) soc.

This board does not include `ST-Link/V2`.
I have `ST-Link/V2` separately.
I connected the `MINI-M4` for `STM32 and ST-Link/V2` as follows.

| ST-Link/V2(Pin no.)   | MINI-M4(Pin no.)    | 
| --------------------- | ------------------- |
| TCK_SWCLK(9)          | PA14(36)            |
| TMS_SWDIO(7)          | PA13(35)            |
| TDI(5)                | PA15(30)            |
| TDO_SWO(13)           | PB3(29)             |
| GND(8)                | GND(31)             |
| NRST(15)              | MCLR(1)             |

[image]

<br/>

And I added the source to work on Zephyr.
        https://github.com/xoduddk123/zephyr/tree/stm32f415rg
[image]

I tested `hello_world` and the command is as follows.
``` bash
source zephyr-env.sh
mkdir -p samples/hello_world/build
cd samples/hello_world/build
cmake -DBOARD=mikroe_mini_m4_for_stm32 ..
make -j8
make flash
```

When I first started flashing, I saw the message below.
``` bash
Error: stm32x device protected
```

So I entered the following command and it worked fine.
``` bash
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f /usr/share/openocd/scripts/target/stm32f4x.cfg -c init -c "reset halt" -c "stm32f4x unlock 0" -c "reset run" -c shutdown
```

`hello world` is output through UART2.
[image]

I have not worked on the document (* .rst) yet.
There is also no LED information in the device-tree yet.
Can I do `Pull request 'if I do more work?

I'm sorry I do not have enough English.
Thank you.

P.s I am a beginner of zephyr and got help from @erwango.(#12568)

erwango, 2019-01-21:
@xoduddk123 , you're welcome to do a pull request.
Please make sure to provide enough documentation on the board so we can check board specifications.

xoduddk123, 2019-01-22:
@erwango Okay.
I will add the `LED` and the `document` and make a `Pull request`.

PR #12951 (2019-04-18): boards: arm: add mikroe_mini_m4_for_stm32
Add support for STM32F415RG.
Add support for [mikroe_mini_m4_for_stm32](https://www.mikroe.com/mini-stm32f4).
CC: @erwango

Fixes #12602
closed 2019-04-18 by galak