zephyrproject-rtos/zephyr #12611: FRDM-K64F + telnet-sample

radfanSC, 2019-01-21:
**Describe the bug**
I have created the example Telnet console application for the board FRDM-K64F under Windows. 
Although I get a connection to the board with a telnet client, but there are no inputs and outputs on the client possible.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/telnet
2. mkdir build; cd build
4. cmake -GNinja -DBOARD=frdm_k64f ..
5. ninja
6. Put the result zephyr/zephyr.bin into the board
7. wait until dhcpv4 has finished
8. connect with a Telnet-client 

**Expected behavior**
Input/Output to the Shell via Telnet-console

**console output**
uart:~$ ***** Booting Zephyr OS zephyr-v1.13.0-2584-g8ec7c73d2e *****
[00:00:00.008,451] <inf> net_config.net_config_init: Initializing network
[00:00:00.008,474] <inf> net_config.setup_ipv4: IPv4 address: 10.0.0.62
[00:00:00.008,489] <inf> net_config.setup_dhcpv4: Running dhcpv4 client...
[00:00:06.021,795] <inf> net_dhcpv4.dhcpv4_handle_msg_ack: Received: 10.0.1.32
[00:00:06.021,829] <inf> net_config.ipv4_addr_add_handler: IPv4 address: 10.0.1.32
[00:00:06.021,832] <inf> net_config.ipv4_addr_add_handler: Lease time: 86400 seconds
[00:00:06.021,846] <inf> net_config.ipv4_addr_add_handler: Subnet: 255.255.254.0
[00:00:06.021,859] <inf> net_config.ipv4_addr_add_handler: Router: 10.0.0.254
[00:00:08.020,315] <dbg> net_telnet.telnet_setup_server: Telnet console enabled on AF_INET
[00:00:08.020,341] <inf> net_telnet.telnet_console_init: Telnet console initialized
[00:00:08.020,346] <inf> net_telnet_sample.main: Starting Telnet sample
[00:00:19.963,597] <dbg> net_telnet.telnet_accept: Telnet client connected (family AF_INET)

**Environment:**
 - OS: Windows
 - Toolchain: gccarmemb Version 7.3.1

jukkar, 2019-01-21:
Tested this also with native_posix, I am not seeing any output from `nc 2001:db8::1 23`.

nordic-krch, 2019-01-30:
Please specify as i'm not sure what is the issue. Shell over uart stops working? Or telnet? afaik, there is no shell over network.

radfanSC, 2019-01-31:
The shell is working over uart but not over telnet.
In the documentation stands:
"You are now connected, and as for the UART console, you can type in your commands and get the output through your telnet Client"

nordic-krch, 2019-01-31:
@jukkar did legacy shell supported network transport? New one doesn't yet, so that would look like a misleading documentation after switching to new shell.

tbursztyka, 2019-01-31:
@nordic-doki that sounds like a bug introduced by new shell. Legacy stuff was working properly, but remember that it was far easier back then: all was using printk below...

agansari, 2019-03-20:
This issue look more like a feature than a bug @galak .
I've tested in qemu and posix, but there is not shell in telnet console. It's not restricted to k64f.

> Yes, the telnet cannot be used atm as the shell is missing support for network transport.
> 
> _Originally posted by @jukkar in https://github.com/zephyrproject-rtos/zephyr/issues/14691#issuecomment-474730959_

jukkar, 2019-03-20:
This is generic shell issue, fixing title.

nordic-krch, 2019-03-26:
Could we reassign to someone else? Adding shell network backend is not in my plans and scope.

PR #15026 (2019-04-01): shell: Add TELNET backend
Introduce new TELNET backend for shell module, based on the `telnet_console` driver. As the orignal code did not fit well with the shell API (it used `printk` and `line_fifo`), I've reused pieces of code and created a new shell backend with it.

As the old `telnet_console` is based on the deprecated `line_fifo` and is on longer used in Zephyr, so I took the liberty and removed it.

There is a little support for TELNET commands (just as in the original code), but I'm aware it's far from ideal.

Resolves #12611.
closed 2019-04-01 by nashif