zephyrproject-rtos/zephyr #12614: [bluetooth] is there any samples I can refer to understand primary or general subnets more precisely?

summer20100514, 2019-01-21:
I read the Mesh Profile V1.0 and I understand that one subnet is only a matter of keys, which are network keys and application keys. Nodes have the same network keys belong to the same subnet. Now I have two questions. Is it true that the mesh network topology is not hierarchical, for there is only one important thing which is the specific key. And what confused me most is that are nodes in different subnets can be addressed at the same time only by virtual address?

jhedberg, 2019-01-21:
I think most samples only use a single subnet. IIRC the mesh_shell one (tests/bluetooth/mesh_shell) might have support for multiple subnets.

For each node, there is a concept of primary network key, which is the key that was given to it as part of the provisioning data (as opposed to something that came through the configuration model). A node will e.g. only react to IV Update changes on the primary network key.

Then you've also got application keys, which are bound to a specific network key, so in that sense I suppose you could think of there being a two-layer hierarchy.

summer20100514, 2019-01-22:
@jhedberg thank you for your detailed explanations. What's more, when and how can I use the virtual address? Is virtual address visible between different subnets?

closed 2019-02-11 by summer20100514