zephyrproject-rtos/zephyr #12624: topic-counter: missing support for channel status

pabigot, 2019-01-21:
There doesn't appear to be any API to determine the status of a specific channel.  The states of a channel may include:

* Available for use
* Configured for an alarm that has not been reached
* Configured for an alarm that has been reached but handler has not been invoked
* Currently invoking the handler
* No such channel

These states become important when trying to coordinate application use of a counter channel with asynchronous processing of compare events.

nordic-krch, 2019-01-22:
`set_channel_alarm` returns `-EBUSY` if channel is occupied. There is a `counter_get_num_of_channels()` function. As for intermediate state (alarm not reached, alarm reach but not invoked, invoking) I am not convinced that it brings value as intermediate states can also be distinguished from within interrupt with higher priority so i don't see use cases for that and that will increase complexity of each implementation.

pabigot, 2019-01-22:
> `set_channel_alarm` returns `-EBUSY` if channel is occupied.

That's not documented in `topic-counters` for `counter_set_ch_alarm()`.  It is documented for `counter_set_wrap()`.

See https://github.com/zephyrproject-rtos/zephyr/issues/12625#issuecomment-456392058 for a use case where status is needed synchronous to other operations.  For completeness it should also be available through a standalone API call.