zephyrproject-rtos/zephyr #12632: tests/drivers/adc/adc_api/ fails on quark platforms

arun1joshi, 2019-01-22:
The test in tests/drivers/adc/adc_api/ fails with Assertion Failure on on quark platforms.

The test is failing after this commit: 0906a51

**Arch**: x86, ARC
**Platforms**: quark_se_c1000, quark_d2000

**Error on console log**:

_quark_se_c1000:arc_
`***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS zephyr-v1.13.0-3318-g0906a51dac (delayed boot 1000ms) *****
 Running test suite adc_basic_test
 ===================================================================
 starting test - test_adc_sample_one_channel

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:212: test_task_one_channel: (ret not equal to 0)
 adc_read() failed with code -5
 FAIL - test_adc_sample_one_channel
 ===================================================================
 starting test - test_adc_sample_two_channels

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:246: test_task_two_channels: (ret not equal to 0)
 adc_read() failed with code -5
 FAIL - test_adc_sample_two_channels
 ===================================================================
 starting test - test_adc_asynchronous_call

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:296: test_task_asynchronous_call: (ret not equal to 0)
 adc_read_async() failed with code -22
 FAIL - test_adc_asynchronous_call
 ===================================================================
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS zephyr-v1.13.0-3318-g0906a51dac (delayed boot 1000ms) *****
 Running test suite adc_basic_test
 ===================================================================
 starting test - test_adc_sample_one_channel

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:212: test_task_one_channel: (ret not equal to 0)
 adc_read() failed with code -5
 FAIL - test_adc_sample_one_channel
 ===================================================================
 starting test - test_adc_sample_two_channels

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:246: test_task_two_channels: (ret not equal to 0)
 adc_read() failed with code -5
 FAIL - test_adc_sample_two_channels
 ===================================================================
 starting test - test_adc_asynchronous_call

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:297: test_task_asynchronous_call: (ret not equal to 0)
 adc_read_async() failed with code -21
 FAIL - test_adc_asynchronous_call
 ===================================================================
 starting test - test_adc_sample_with_interval
 sample_with_interval_callback: sampling 1
 sample_with_interval_callback: sampling 2
 sample_with_interval_callback: sampling 3
 sample_with_interval_callback: sampling 4
 sample_with_interval_callback: sampling 5

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:351: test_task_with_interval: (ret not equal to 0)
 adc_read() failed with code -4
 FAIL - test_adc_sample_with_interval
 ===================================================================
 starting test - test_adc_repeated_samplings

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:438: test_task_repeated_samplings: (ret not equal to 0)
 adc_read() failed with code -21
 FAIL - test_adc_repeated_samplings
 ===================================================================
 starting test - test_adc_invalid_request

     Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:484: test_task_invalid_request: (ret not equal to 0)
 adc_read() failed with code -4
 FAIL - test_adc_invalid_request
 ===================================================================
 Test suite adc_basic_test failed.
 ===================================================================
 PROJECT EXECUTION FAILED`

On _quark_d2000:x86_, only 2 test cases fail - test_adc_asynchronous_call, test_adc_repeated_samplings:
`found expected `\*\*\*\*\* Booting Zephyr OS .* \(delayed boot [0-9]+ms\) *\*\*\*\*\*` in console `srrsotc04/QA-mv-01:default` at 1.30s
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-3318-g0906a51dac (delayed boot 1000ms) *****
Running test suite adc_basic_test
===================================================================
starting test - test_adc_sample_one_channel
Samples read: 0x03e4 0x0000 0x0000 0x0000 0x0000 0x0000
PASS - test_adc_sample_one_channel
===================================================================
starting test - test_adc_sample_two_channels
Samples read: 0x03ff 0x03fe 0x0000 0x0000 0x0000 0x0000
PASS - test_adc_sample_two_channels
===================================================================
starting test - test_adc_asynchronous_call
console output:
    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:296: test_task_asynchronous_call: (ret not equal to 0)
adc_read_async() failed with code -22
FAIL - test_adc_asynchronous_call
===================================================================
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-3318-g0906a51dac (delayed boot 1000ms) *****
Running test suite adc_basic_test
===================================================================
starting test - test_adc_sample_one_channel
Samples read: 0x03e4 0x0000 0x0000 0x0000 0x0000 0x0000
PASS - test_adc_sample_one_channel
===================================================================
starting test - test_adc_sample_two_channels
Samples read: 0x03ff 0x03fe 0x0000 0x0000 0x0000 0x0000
PASS - test_adc_sample_two_channels
===================================================================
starting test - test_adc_asynchronous_call
console output:
    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:296: test_task_asynchronous_call: (ret not equal to 0)
adc_read_async() failed with code -22
FAIL - test_adc_asynchronous_call
===================================================================
starting test - test_adc_sample_with_interval
sample_with_interval_callback: sampling 0
sample_with_interval_callback: sampling 1
sample_with_interval_callback: sampling 2
sample_with_interval_callback: sampling 3
sample_with_interval_callback: sampling 4
Samples read: 0x03fc 0x03fd 0x03fb 0x03fe 0x03ff 0x0000
PASS - test_adc_sample_with_interval
===================================================================
starting test - test_adc_repeated_samplings
console output:
    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:437: test_task_repeated_samplings: (ret not equal to 0)
adc_read() failed with code -22
FAIL - test_adc_repeated_samplings
===================================================================
starting test - test_adc_invalid_request
Samples read: 0x03ff 0x0000 0x0000 0x0000 0x0000 0x0000
PASS - test_adc_invalid_request
===================================================================
Test suite adc_basic_test failed.
===================================================================
PROJECT EXECUTION FAILED`

cinlyooi-intel, 2019-02-19:
Slightly different output from quark_se_c1000_ss_devboard:arc for commit cf94d6811aeb6cf345c678862729ecb1e91fc272. Extra Assertion failures. I did not check adc_read() return code

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-264-gcf94d68 (delayed boot 1000ms) *****
Running test suite adc_basic_test
===================================================================
starting test - test_adc_sample_one_channel

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:214: test_task_one_channel: (ret not equal to 0)
adc_read() failed with code -5
FAIL - test_adc_sample_one_channel
==========================================
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-264-gcf94d68 (delayed boot 1000ms) *****
Running test suite adc_basic_test
===================================================================
starting test - test_adc_sample_one_channel

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:214: test_task_one_channel: (ret not equal to 0)
adc_read() failed with code -5
FAIL - test_adc_sample_one_channel
===================================================================
starting test - test_adc_sample_two_channels

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:248: test_task_two_channels: (ret not equal to 0)
adc_read() failed with code -5
FAIL - test_adc_sample_two_channels
===================================================================
starting test - test_adc_asynchronous_call

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:298: test_task_asynchronous_call: (ret not equal to 0)
adc_read_async() failed with code -22
FAIL - test_adc_asynchronous_call
===================================================================
starting test - test_adc_sample_with_interval
sample_with_interval_callback: sampling 0
sample_with_interval_callback: sampling 1
sample_with_interval_callback: sampling 2
sample_with_interval_callback: sampling 3
sample_with_interval_callback: sampling 4

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:352: test_task_with_interval: (ret not equal to 0)
adc_read() failed with code -5
FAIL - test_adc_sample_with_interval
===================================================================
starting test - test_adc_repeated_samplings

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:439: test_task_repeated_samplings: (ret not equal to 0)
adc_read() failed with code -22
FAIL - test_adc_repeated_samplings
===================================================================
starting test - test_adc_invalid_request

    Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/drivers/adc/adc_api/src/test_adc.c:485: test_task_invalid_request: (ret not equal to 0)
adc_read() failed with code -5
FAIL - test_adc_invalid_request
===================================================================
Test suite adc_basic_test failed.
===================================================================
```

cinlyooi-intel, 2019-02-19:
quark_d2000_crb:x86 for commit cf94d6811aeb6cf345c678862729ecb1e91fc272 has same failures

nashif, 2019-03-05:
@dcpleung you had dealt with the ADC for quark recently, can you please take a look?

cinlyooi-intel, 2019-03-15:
Reproduce  on

1. quark_d2000:x86, and
2. quark_se_c1000:arc

for Commit ID: 96235c5027461534b75059a65f8833231d293312

closed 2019-03-21 by andrewboie

arun1joshi, 2019-03-28:
verified ok on commit [cf91f8cf90]

commit 71b4137d (2019-03-21): adc: overhaul adc_dw and rename it to adc_intel_quark_se_c1000_ss

Since commit 0906a51dacc947c5fa19a80e208549f9039556cd, the driver
fails the test: tests/drivers/adc/adc_api. Some of the workflow
turns out to be incorrect (e.g. not doing dummy conversion after
getting out of deep power down, and not clearing interrupt status
bits). So take some time to overhaul the driver. Also rename
the driver to adc_intel_quark_se_c1000_ss because the inner
working of this driver is tied to Quark SE C1000 SoC.

Fixes: #12632
commit 86e47383 (2019-03-21): adc: quark_d2000: fix sampling conversion

The logic in sampling is incorrect. The code sets up the hardware
to do multiple conversions, but in reality, each call is simply
one conversion. So fix it.

Fixes #12632