zephyrproject-rtos/zephyr #12635: tests/subsys/fs/nffs_fs_api/common/nffs_test_utils.c fail with Assertion failure on nrf52840

arun1joshi, 2019-01-22:
The tests in tests/subsys/fs/nffs_fs_api/common/nffs_test_utils.c fail with Assertion failure on nrf52840

The test is failing after this commit: f0120848244be7d88f0b831c76e9e2da43f7dbde

**Arch**: arm
**Platforms**: nrf52840_pca10056

**Error on console log**: 
_nrf52840_pca10056:arm_
```

***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-3391-gf012084824 (delayed boot 1000ms) *****
Running test suite nffs_fs_large_test
===================================================================
starting test - test_fs_mount
PASS - test_fs_mount
===================================================================
starting test - test_large_unlink
Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/subsys/fs/nffs_fs_api/common/nffs_test_utils.c:251: nffs_test_util_create_file_blocks: rc not equal to blocks[i].data_len
FAIL - test_large_unlink
===================================================================
starting test - test_large_system
Assertion failed at /home/jenkins/workspace/zephyr-master-ci-master/zephyr.git/tests/subsys/fs/nffs_fs_api/common/nffs_test_utils.c:460: nffs_test_assert_branch_touched: i < nffs_test_num_touched_entries is false
FAIL - test_large_system
===================================================================
Test suite nffs_fs_large_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

nvlsianpu, 2019-01-31:
I'm investigating this.

cinlyooi-intel, 2019-02-27:
Please note that there are a lot of `<err> fs: failed to unlink path (-2)&#1b;[0m` messages after `PROJECT EXECUTION FAILED`

nvlsianpu, 2019-02-28:
> Please note that there are a lot of `<err> fs: failed to unlink path (-2)&#1b;[0m` messages after `PROJECT EXECUTION FAILED`

Test is designed in such a manner that some of this unlink warning are expected. See https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/subsys/fs/nffs_fs_api/common/nffs_test_utils.c#L238

This allso have to be fixed, as it misleading bug invastigation.

nvlsianpu, 2019-02-28:
fix in progress fork branch: https://github.com/nvlsianpu/zephyr/tree/bugfix/nffs_large_tests

nvlsianpu, 2019-03-01:
After investigation `test_large_system` -  fails after GC and FS flash re-parsing . It is caused by ["NFFS can store a valid file-system instance which is unrecoverable after reset." bug](https://github.com/apache/mynewt-nffs/issues/10), so it is also manifestation of https://github.com/zephyrproject-rtos/zephyr/issues/9748 error source.

nvlsianpu, 2019-03-26:
I lowered priority as NFFS is external project and I know that jullabs (NFFS project owner) is working on the NFFS fixing but this will be not ready soon.

nvlsianpu, 2019-03-26:
cc @carlescufi

carlescufi, 2019-03-26:
@nvlsianpu understood, thank you

nvlsianpu, 2020-01-27:
NFFS support was removed #21793.

closed 2020-01-27 by nvlsianpu