zephyrproject-rtos/zephyr #12652: UART console is showing garbage with driver uart_ns15560

franciscomunoz, 2019-01-23:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
 Removed the + 1 in the following line
               divisor = ((dev_cfg->sys_clk_freq / baud_rate) >> 4) + 1;
and characters are properly printed.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD= "The board is not in the tree yet"
3. make
4. See error

**Expected behavior**
Data is not properly printed in the UART console

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.
\\\
[image]

\\\
**Environment (please complete the following information):**

SHA:
               commit f0db0b8815a49c6bc5c489072199d38876298f8a
              Author: Wayne Ren <wei.ren@synopsys.com>
              Date:   Sun Nov 18 02:21:09 2018 +0800

nashif, 2019-01-23:
@vonhust I did not spot those changes to the driver when it was submitted. We should not have board/soc specific changes in the driver, if the driver requires some other values it should be supported as a new kconfig.
Can you please remove the IOTDK related checks and instead make this configurable in some other way (using kconfig for example)?

franciscomunoz, 2019-01-23:
The selection of UART_REG_ADDR_INTERVAL is also wrong. We have a different ARM SOC which requires UART_REG_ADDR_INTERVAL to be 1.
The patch from @fabled is far better :

https://github.com/zephyrproject-rtos/zephyr/pull/5383/commits/c1e41d41f8dcebaf54a6a417d295bf5c480891e3

vonhust, 2019-01-29:
@franciscomunoz @nashif  I understand it's better not to add SoC specific code in driver, but the code i added should just work for ARC_IOT,  should not impact other target.  

For "the divisor = ((dev_cfg->sys_clk_freq / baud_rate) >> 4) + 1",  it works for emsk which also uses ns16550 compatble DW uart IP.   If i remove the "+1", the iotdk dost not output correctly.

vonhust, 2019-01-29:
@franciscomunoz @nashif  The correct baud caculation formula should be
`divisor` = (u32_t)(dev_cfg->sys_clk_freq / (16.0 * baud_rate) + 0.5);' . The original one 
'divisor = (dev_cfg->sys_clk_req/ baud_rate) >> 4' does not work for iotdk whose sys_clk is 16 Mhz. whe baud_rate is 115200,  the baudrate error is out of limit. 

However,  if apply `divisor` = (u32_t)(dev_cfg->sys_clk_freq / (16.0 * baud_rate) + 0.5);' , is it ok to bring float (SW/HW) caculation in kernel?  
If not apply the new formual,  the original way can not meet all the possible combinations of sclk and baudrate.

nashif, 2019-01-29:
probably related: #12810

dcpleung, 2019-01-30:
#12810 is also due to the "+ 1" in divisor calculation.

dcpleung, 2019-01-30:
@vonhust The PR #12807 works on the UP Squared board.

PR #12807 (2019-01-31): drivers: uart_ns16550: remove soc specific codes and bug fixes
* remove soc specific codes
* optimize the caculation of baudrate.
   the correct formula is 
        divisor =  (u32_t)(dev_cfg->sys_clk_freq / (16.0 * baud_rate) + 0.5)
  here a variant is used to replace the old one

Fixes #12652
closed 2019-01-31 by nashif