zephyrproject-rtos/zephyr #12677: USB: There are some limitations for users to process descriptors

HelloOumiga, 2019-01-24:
When I am doing some development related to the Zephyr USB device stack, I  found that the descriptors process has the following limitations:
1.  The class interface descriptors are in the class driver (like \subsys\usb\class\hid\core.c ), the existed descriptors are only aimed for one instance, sometimes I need to enable a composite device, for example HID+HID.
2.  For the high speed device, it should provide different MPS/interval when it connects to HS/FS host. MPS/interval should be updated according to the current speed mode. Currently, there is no way to provide the different descriptor accordingly. 
3. For customized endpoint number, there is no way for the application to specify the endpoint number to a special function in the special user case. I think this is an issue.

finikorg, 2019-01-30:
I have PR #12682 making it possible to have several HID instances.

finikorg, 2019-01-30:
@HelloOumiga About
```
For customized endpoint number, there is no way for the application to specify the endpoint number to a special function in the special user case. I think this is an issue.
```
how do you want to use endpoint? I think in this case you want to describe descriptor table yourself and add all callbacks, look to wpanusb or webusb samples.

finikorg, 2019-01-31:
Do you mean that you want to assign specific endpoint address to each endpoint? Look at wpanusb sample.

HelloOumiga, 2019-01-31:
Yes, this is my meaning. I have found the webusb sample not wpanusb sample,

finikorg, 2019-02-01:
@HelloOumiga I believe endpoint addresses should be assigned automatically, otherwise if you include several instances of HID (mouse and keyboard) or HID + CDC ACM how do you assign those addresses?
I think this is not what you want either.
We are not going to move class code to main, this was old interface, but if you want some very specific device you can do this in your sample.

jfischer-phytec-iot, 2019-02-04:
>For customized endpoint number, there is no way for the application to specify the endpoint number to a special function in the special user case. I think this is an issue.

No is not. a) if you interface has a endpoint there will be endpoint descriptor, you host driver should use it and obtain the endpoint address from there. b)on device, the default endpoint address may be updated after stack initialization.

jfischer-phytec-iot, 2019-09-16:
closed, see comments above

closed 2019-09-16 by jfischer-phytec-iot