zephyrproject-rtos/zephyr #12681: BLE Split Link Layer

carlescufi, 2019-01-24:
List of tasks remaining to consider the new `CONFIG_BT_LL_SW_SPLIT` Link Layer complete:

- [x] Merge the new Link Layer to master
- [ ] Add support for Data Length Extensions
- [ ] Add support for Controller-based Privacy
- [ ] Add support for Mesh Extensions
- [ ] Use `atomic_cas` when supported by the arch without disabling interrupts ([link](https://github.com/zephyrproject-rtos/zephyr/pull/12404#pullrequestreview-192102680))
- [ ] Document all primitives (@mped-oticon)

carlescufi, 2019-01-24:
CC @jhedberg @Vudentz

aescolar, 2019-01-24:
CC @mtpr-ot @thoh-ot  @wopu-ot

aescolar, 2019-01-24:
@pskhansen

jhedberg, 2019-01-24:
Does the new controller support Advertising Extensions "out of the box"? If not, that should be added to the list as well, I suppose.

pskhansen, 2019-01-24:
@carlescufi A few more tasks for the list:
- Change the role dependent event-done handling from type switch-case to callback concept.
- Move the LLL event prepare/resume queue/handling into LLL.
- Move the LLL-to-ULL Tx ack queue API to ULL domain.

jhedberg, 2019-01-24:
> @jhedberg it doesn't, but this list is really about reaching feature parity with the existing Link Layer. We already have a GitHub issue for AE.

@carlescufi ok, thanks for clarifying. I think "Add support for Mesh Extensions" confused me, since that's not really (at least fully) supported in the old LL either.

aescolar, 2019-01-24:
> Out of curiosity, is "remove old link layer" in the road map, or is the intention to support both?

@SebastianBoe : Eventually yes, but the 1st step is to get the split LL to match its functionality

koffes, 2019-02-08:
@carlescufi @cvinayak if possible, please break down some more of the issues defined here in smaller tasks. Encapsulate the `req/ack` in PR #12681 was a nice bite size for getting started.

koffes, 2019-02-14:
FYI: I am currently implementing atomic_cas as discussed above.

PR #15781 (2019-07-22): Bluetooth: controller: Make ULL/LLL split the default
Make the Upper and Lower Link Layer split architecture
implementation of the controller as the default when
building Zephyr Bluetooth Low Energy controller support
in applications.

Noticeable missing feature (porting) in comparison to old
architecture implementation is, Advanced scheduling of
connection events.

The missing features will subsequently be submitted
upstream.

Fixes #12681
Depends on #16691
closed 2019-07-22 by carlescufi