zephyrproject-rtos/zephyr #12698: Revise SPI API to move chip-select content from subordinate device to bus

pabigot, 2019-01-24:
Per dev review telecon 2019-01-24 it's agreed that the resolution of #12226 is that the Linux device tree standard of defining chip selects as a `cs-gpios` property in the bus node will be followed.  For consistency this means that we should change the existing Zephyr SPI api and driver implementation so that chip-select management is performed by the bus rather than in each subordinate device driver implementation.

Without careful review this might mean something like removing the `spi_cs_control` pointer from `spi_config` and providing API on the parent bus to assert and deassert the chip-select identified by the the `slave` field.

tbursztyka, 2019-01-24:
>  For consistency this means that we should change the existing Zephyr SPI api

What? Just because linux does its way, zephyr should do the same? No, definitely no. Plus, the way DT does its thing (the linux way) does not generate any issue for the implemented zephyr way. There was a good reason for putting the cs logic into spi device and not into spi master. As for not providing function to assert/deassert the slave line.

pabigot, 2019-01-24:
The architectural decision to follow Linux is isolated to device tree content.  The question here is the impact of that decision on the SPI API.

Currently every subordinate device driver must define and initialize the `spi_cs_control` structure for their device when the device is used on a target where CS is controlled by GPIO rather than a dedicated bus signal.  Structurally the bus driver should be responsible for those definitions.

This is an issue so we can discuss it in the open, determine when it's going to happen, and who's going to do it.  I suspect it can mostly be limited to `spi_context.h`.

tbursztyka, 2019-01-25:
>  The question here is the impact of that decision on the SPI API.

None.

erwango, 2019-01-25:
My concern on Linux way of defining cs-gpio is, in case of multiple slaves on the bus, how can a driver knows which cs-gpio controls a defined slave ?
For instance, in disco_l475_iot1 example:
```
&spi3 {
	status = "ok";

	cs-gpios = <&gpiod 13 0>, <&gpioe 0 0>;

	bt0: spbtle-rf@0 {
		compatible = "st,spbtle-rf";
                [...]
		label = "SPBTLE-RF";
	};

	wifi0: ism43362@1 {
		compatible = "inventek,eswifi";
                [...]
		label = "ESWIFI0";
	};
};

```
I guess rule here is index of cs-gpio matches index of slave in the list.
This seems the logical way, but it could be confusing to non dts aware users.
By the way https://github.com/zephyrproject-rtos/zephyr/pull/12647 is wrong here as it selects the wrong cs-gpio for "ESWIFI0". I'll raise an issue for that.

Then, how will it happens in case of overlays?
Is dtc capable of merging overlays so that overlay cs-gpio is appended to the bus cs-gpio at the same index than the overlay slave is appended to the bus slaves list?

pabigot, 2019-01-25:
@erwango Several of your points were raised and dismissed in #12226.  We can revisit them, but the need to do so continues to highlight Zephyr's fragile and ad-hoc decision-making process.

I believe for overlays the solution is an upstream enhancement request from @galak to device tree that allows overlays to manipulate array property values.  The question of how that can be reconciled with the requirement that both the `reg` property and unit address match the index for the corresponding chip-select is a very good one.