zephyrproject-rtos/zephyr #12703: how to configure interrupt signals on shields via device tree?

pabigot, 2019-01-24:
I've got patches related to the HTS221 sensor for nrf52_pca20020 to enable interrupt-driven sampling.  Works fine, but I'd like to test it with the x_nucleo_iks01a2 shield on `frdm_k64f` and `nucleo_l476rg`.  The corresponding `DRDY` signal is on the Arduino header at `GPIO2`.

Is there an existing solution for this?
```
        hts221@5f {
                compatible = "st,hts221";
                reg = <0x5f>;
                drdy-gpios = <&what goes here>;  /* or &arduino_int0? */
                label = "HTS221";
        };
```

erwango, 2019-10-23:
@pabigot, can we close this point?

pabigot, 2019-10-23:
Yes, I think the nexus node capability handles this.

Wonder if I still have those patches anywhere....

closed 2019-10-23 by pabigot