zephyrproject-rtos/zephyr #12708: net_pkt_(un)ref is used in an unsafe way

dgloeck, 2019-01-24:
**Describe the bug**
In its current form net_pkt_ref and net_pkt_unref are not thread safe. But there are cases where these functions are used on the same packet from different contexts at the same time. If this happens, the reference count of a packet ends up being wrong. In my case net_pkt_unref is called both from `ethernet_send` and from eth_sam_gmac.c's `tx_complete`.

**To Reproduce**
Steps to reproduce the behavior:
1. Send lots of ethernet packets on a SAM E70 board.
2. After several hours you will run out of tx buffers.

**Expected behavior**
The reference count is always correct and sent packets are freed.

**Impact**
None, because I will soon make a pull request for a fix.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK,
 - Commit SHA: 5d91118672cb00bb02146aa6ac7e3588762db737

pfalcon, 2019-01-24:
cc @tbursztyka

pfalcon, 2019-01-24:
@agross-linaro : How do you like this one? ;-)

aurel32, 2019-01-24:
It should be noted that when CONFIG_PTP_CLOCK_MCUX=y the MCUX Ethernet driver also keeps a reference to packet with net_pkt_ref() and later call net_pkt_unref() from the ISR.

aurel32, 2019-01-24:
The SAM E70 driver currently keep a reference to the full packet net_pkt_ref(). It seems this is required by the the PTP part only. Otherwise it seems that we don't need to keep the whole packet, but only the individual fragments, therefore without PTP support we can call net_buf_ref() and net_buf_unref() instead of doing that at the packet level. That might make things easier to fix. Actually we just need to keep the data, not even the link between the various fragments.

agross-linaro, 2019-01-24:
Seems like this might explain some of the TCP header null stuff.

tbursztyka, 2019-01-25:
@dgloeck I don't know what I was thinking this morning, but yes indeed that's definitely a valid issue. (I think I got stuck on unref call orders or on my pillow, either way.)

jukkar, 2019-01-25:
@pfalcon why did you assigned this issue to me as there was already a fix proposal for this this? This is to be  assigned to @dgloeck who submitted the fix proposal.

closed 2019-01-29 by jukkar

pfalcon, 2019-01-30:
> Otherwise it might happen that two persons might work to fix same issue,

That's exactly the reason I assign all networking bugs to you - there should be someone who's in loop on *all* things networking, and as a lead of networking team, that's gotta be you. Assigning to you makes sure you receive notifications of further changes to the issue, even if you reassign it to someone else.

> In this case creating this issue was a bit pointless as there was already a fix, thus creating a bug report is not really needed (IMHO).

Submitting bug tickets is again a measure to preclude cases like you describe above - there should be ticket, with cc: and discussion before work actually commences. This precludes duplication of effort and xy-fixes (a-la http://xyproblem.info/).

agross-linaro, 2019-01-30:
@pfalcon In most cases, the bug gets triaged and assigned to someone to work, especially for well manned subsystems.  Unless you've already talked to someone and they say assign it to me, so long as the tagging on the patch is done correctly, it'll get assigned fairly quickly.

As for duplication of effort, I would expect most people don't scan the list of PRs and Issues to see if there is something possibly the same or related.  I know I don't.  And we don't always have issues before PRs or a issue for every PR.

pfalcon, 2019-01-30:
@agross-linaro 

> it'll get assigned fairly quickly.

I'm not sure what you mean by "it'll get assigned". It won't "get assigned", some specific people will assign it. And all these years, I was among people who was triaging bugs, in a project with ~290 ever-open pull requests, and ~900 open bugs, counts always growing, never going down.

> I know I don't. 

I do. I have to. It's part of the responsibilities of being full-time on a project. But I of course know that many people don't/can't, that's why I'm making sure I @-notify all parties which I know may be involved with a particular matter.

PR #12709 (2019-01-29): net: pkt: fix race condition in packet reference counting
It has been observed that some network drivers, f.ex. the SAM E70 GMAC,
call net_pkt_unref from inside the interrupt that signals the successful
transmission of a packet. This conflicts with the net_pkt_unref call made
by ethernet_send after the packet has been given to the driver.

We fix this by using an atomic_t to hold the reference count as there
might be other, difficult to find cases of net_pkt_(un)ref being used
across threads and interrupts.

The name of the element has been changed from "ref" to "atomic_ref" to
cause a compile [... cut]