zephyrproject-rtos/zephyr #12714: Virtual UART over BLE

tautologyclub, 2019-01-24:
I'd like to create a virtual UART using BLE, i.e. a Zephyr driver implementing the UART API but internally tunnels RX/TX to one or two GATT characteristics (not sure if you need separate characteristics for RX/TX for full duplex or if that's just overcomplicating things)

The obvious use case is a creating a wireless FTDI dongle (sweet). But I can see it being useful beyond this as well: BLE as a logging/console/shell backend wouldn't need any lines of codes at all to implement for instance, just point it to the virtual UART. 

(also you could use this to create an infinite loop where the device speaks HCI/UART with itself, possibly leading to machine sentience in the mid- to long term)

But again, all of this seems almost too obvious, so if there's a sample hidden in the tree that I haven't found yet, I'd be grateful to hear about it.

thelittlebug, 2020-11-21:
@tautologyclub have you found any glues in the meantime?

markus-becker-tridonic-com, 2020-11-25:
Is https://github.com/nrfconnect/sdk-nrf/blob/master/samples/bluetooth/shell_bt_nus/README.rst  what you are looking for?

pvonmoradi, 2020-11-25:
@markus-becker-tridonic-com @tautologyclub 
There is proprietary NUS with samples here: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/bluetooth/peripheral_uart/README.html#  
My problem is, I want to use it from the regular upstream Zephyr (so I don't need to install NCS). Is it possible?

markus-becker-tridonic-com, 2020-11-25:
> @markus-becker-tridonic-com @tautologyclub
> There is proprietary NUS with samples here: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/bluetooth/peripheral_uart/README.html#

That was the sample I meant.

> My problem is, I want to use it from the regular upstream Zephyr (so I don't need to install NCS). Is it possible?

Well, the NCS sample might or might not be restricted to Nordic hardware. YMMV.

pvonmoradi, 2020-11-25:
@markus-becker-tridonic-com NUS *is* restricted to nordic hw. My problem is, I don't want to manage NCS and upstream zephyr at the same time.

cederom, 2021-10-18:
Why still until today there is no SHELL over BLE UART in Zephyr upstream? :-(

kartben, 2024-02-09:
> Why still until today there is no SHELL over BLE UART in Zephyr upstream? :-(

#68076

PR #69881 (2024-03-22): serial: Add UART over Bluetooth LE 
### Description
This PR introduces a UART-compatible implementation of Bluetooth LE GATT, using NUS (Nordic UART Service), which allows users to utilize Bluetooth LE as serial endpoints with UART APIs and take advantage of existing support of subsystems that already support UART.

### Background
This PR fixes #12714. Also, this PR is a follow-up of #68076, with the agreed [changes captured on the Bluetooth WG](https://github.com/zephyrproject-rtos/zephyr/pull/68076#issuecomment-1945971326).

### Changes
- Add NUS as a Bluetooth LE GATT Service.
- Add Sample  [... cut]
closed 2024-03-22 by carlescufi