zephyrproject-rtos/zephyr #12718: Generic MbedTLS setup doesn't use MBEDTLS_ENTROPY_C

mike-scott, 2019-01-24:
**Is your feature request related to a problem? Please describe.**
Seems like we should be setting up MbedTLS's entropy source?  I noticed this as I was moving code from the net_app layer which does include the entropy setup (and thus can't be used with the generic MbedTLS setup).

**Describe the solution you'd like**
Add MBEDTLS_ENTROPY_C to config-tls-generic.h and probably add hooks to:
mbedtls_entropy_init
mbedtls_entropy_add_source

**Describe alternatives you've considered**
None

pfalcon, 2019-01-25:
My understanding that instead of using mbedTLS "entropy subsystem", we use Zephyr "entropy subsystem". So, we don't configure mbedTLS' one with mbedtls_entropy_add_source(), but feed Zephyr's entropy source directly into the next mbedTLS layer in order (which is CTR_DRBG):

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/sockets/sockets_tls.c#L280 . dev there is Z entropy device.

pfalcon, 2019-01-25:
cc @d3zd3z

jukkar, 2019-02-21:
There are too many assignees to this issue so no one is probably looking this. I remember @rlubos doing something with mbedtls last time so assigning to him. If you cannot work with this, please assign back to me.

As this is a feature request, I am removing the 1.14 milestone as the merge window is already closed.

rlubos, 2019-02-21:
As mentioned earlier, we use Zephyr entropy subsystem directly in secure sockets, so I'm not sure if there's an actual value in piping it through mbedTLS entropy module.

Regardless of the secure sockets, adding an option to enable mbedTLS entropy is a good idea. I actually need this change to resolve #3877, so I'll push a PR once we're done with the release.

d3zd3z, 2019-12-02:
It is important to distinguish between entropy and a pseudorandom source. The output of the entropy subsystem is unsuitable for pretty much anything other than feeding a PRBG. Does secure sockets have its own PRBG that is uses? Otherwise, this setup is not really secure.

Some work by @dleach02 made this distinction clearer, providing a random API in Zephyr that uses an entropy source under the hood.

rlubos, 2019-12-03:
Secure sockets use mbedTLS' CTR-DRBG generator, which is configured to use Zephyr entropy subsystem as an entropy source.

By `we use Zephyr entropy subsystem directly` I meant that we do not pipe it through mbedTLS' entropy module (which to my understanding can collect entropy from multiple sources), not that we use it as a RNG.

rlubos, 2021-02-24:
A Kconfig option to enable `MBEDTLS_ENTROPY_C` in config-tls-generic.h was added in https://github.com/zephyrproject-rtos/zephyr/pull/15648 and I think it was well justified that there's no need to use mbedTLS entropy module in the secure sockets layer, so I think we can close this issue. Please reopen if you disagree.

closed 2021-02-24 by rlubos

mglettig, 2021-05-11:
Hi @rlubos 
If you have a look at the latest release of civetweb that has the option to use mbedtls for HTTPS it affects the discussion here. Please have a look at the following function: https://github.com/civetweb/civetweb/blob/master/src/mod_mbedtls.inl#L57

This needs MBEDTLS_ENTROPY_C to be enabled. But if I enable it it leads to a runtime fault because no entropy source was added in Zephyr. Shouldn't Zephyr call mbedtls_entropy_add_source if CONFIG_MBEDTLS_ENTROPY_ENABLED is set?

This question has to do with #34438.

rlubos, 2021-05-11:
I don't think Zephys as a OS can call `mbedtls_entropy_add_source()` for civetweb, it requires a pointer to `mbedtls_entropy_context` structure, which seems to be defined internally somewhere in the civetweb.

One thing that I believe Zephyr could do is to provide a default entropy source (sorry if I mix up, I don't have much experience with mbedTLS entropy). Currently Zephyr's config file for mbedtls defines the following:
```
#define MBEDTLS_NO_PLATFORM_ENTROPY
#define MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES
```
If those are removed, Zephyr could implement a default `mbedtls_platform_entropy_poll()` function to serve as an entropy source. It shouldn't be enforced though, i. e. the user should still be able to disable the default source with Kconfig and provide his own entropy implementation.