zephyrproject-rtos/zephyr #12719: DISCARD .note.GNU-stack sections on -m32 x86 to silence numerous linker warnings

marc-hb, 2019-01-25:
Building x86 with ZEPHYR_TOOLCHAIN_VARIANT=host (fixed by PR #12674) prints dozens of .note.GNU-stack warnings like this one:

```
/usr/bin/ld: warning: orphan section `.note.GNU-stack' from `drivers/spi/libdrivers__spi.a(spi_intel.c.obj)' being placed in section `.note.GNU-stack'.
/usr/bin/ld: warning: orphan section `.note.GNU-stack' from `drivers/serial/libdrivers__serial.a(uart_ns16550.c.obj)' being placed in section `.note.GNU-stack'.
/usr/bin/ld: warning: orphan section `.note.GNU-stack' from `kernel/libkernel.a(device.c.obj)' being placed in section `.note.GNU-stack'.
/usr/bin/ld: warning: orphan section `.note.GNU-stack' from `kernel/libkernel.a(errno.c.obj)' being placed in section `.note.GNU-stack'.
```

Quoting @nashif in (unrelated) PR #12674:
> @marc-hb We can discard most of those, they are already being discarded for x86_64 so I think we need to put the discard in a more common place.

Previous `arc/v2/linker.ld` example of what I believe @nashif refers to: PR #12524 (couldn't find where it's done for x86_64)

One pretty good introduction to executable stacks, .note.GNU-stack and the corresponding security considerations: https://www.airs.com/blog/archives/518

marc-hb, 2019-01-25:
The .note.GNU-stack section is merely a message from the compiler to the linker, it's never meant to be placed anywhere so this warning looks like a linker bug to me. I suspect using a linker script bypasses the linker logic that normally deals with these sections and removes them after it's done. @ruuddw  any idea?

While the regular .note.GNU-stack logic is well documented (see for instance  https://www.airs.com/blog/archives/518), I'm curious what happens when using a linker script? I mean how does GNU ld make its executable stack decision when using a linker script?

ruuddw, 2019-01-25:
See also my comment in PR #12524 .
The .note.GNU-stack isn't a real section, it's size is always 0. By means of the section attributes it indicates if the code generated by the compiler includes trampoline code written to and executed from the stack.
Not sure how the linker processes this information, I guess it aggregates the info and indicates the result for a linker assigned stack segment, if any. In case of Linux, the elf loader can use it to set protections at runtime. For Zephyr, things work differently as there is no runtime elf loader. If an architecture allows for it, stacks are/should be set to non-executable. The elf protection attributes are not used at all.

marc-hb, 2019-01-25:
FWIW my current and ugly workaround is to turn off LINKER_ORPHAN_SECTION_WARN in Kconfig.zephyr

ruuddw, 2019-01-25:
That is a brute force solution, we might lose relevant warnings as well in the future. The orphan warnings were introduced for a reason (#5534).

Why not adding below to the linker file?
`    /DISCARD/ : { *(.note.GNU-stack) }`
I see @nashif 's concern was duplicating this over different architectures instead of moving it to a common place. Guess it can be added to the common linker/include, like rel-sections.ld and debug-sections.ld

marc-hb, 2019-01-25:
> For Zephyr, things work differently as there is no runtime elf loader

Runtime ELF loader aside, the build system could in theory still rely on the usual and elaborate .note.GNU-stack linking logic, let it run and consult its valuable conclusion in the GNU_STACK program header of the zephyr*.elf file before making its own, final decision(s). No? For instance that would detect *at build time* a misconfiguration where someone disables execstack but uses some nested function anyway.

> If an architecture allows for it, stacks are/should be set to non-executable.

Any example from the top of your head?

closed 2019-01-26 by nashif

ruuddw, 2019-01-28:
>>If an architecture allows for it, stacks are/should be set to non-executable.

>Any example from the top of your head?

As an example, see the mpu code for ARC. Attributes for stack region are set to R/W (not executable).
arch/arc/core/mpu/arc_mpu.c:
`	case THREAD_STACK_REGION:
		return  AUX_MPU_RDP_KW | AUX_MPU_RDP_KR;`

PR #12752 (2019-01-26): linker: x86: add orphan linker sections
Add missing linker section to avoid warning about orphans when building
with host compiler.

Fixes #12719