zephyrproject-rtos/zephyr #12726: Dead loop of the kernel during Bluetooth Mesh pressure communication

xiaoliang314, 2019-01-25:
**Describe the bug**
I created two Bluetooth Mesh nodes to put them in the same network. I use the serial to control one of the nodes to send packets to another node. Send the next packet immediately after receiving a reply from another node or a 5 second timeout. When I tested about 800 times, the node could not continue to respond to my serial commands. I used gdb to check the current running status. The information is as follows.

**To Reproduce**
Steps to reproduce the behavior:
1. cmake -DBOARD="nrf52_pca10040"  ../..
2. make
3. See error

**gdb debug log**
(gdb) 
_priq_dumb_add (pq=pq@entry=0x200019ac <_kernel+40>, 
    thread=thread@entry=0x2000064c <adv_thread_data>)
    at /home/ubuntu/zephyr/kernel/sched.c:574
574		SYS_DLIST_FOR_EACH_CONTAINER(pq, t, base.qnode_dlist) {
(gdb) l
569	{
570		struct k_thread *t;
571	
572		__ASSERT_NO_MSG(!_is_idle(thread));
573	
574		SYS_DLIST_FOR_EACH_CONTAINER(pq, t, base.qnode_dlist) {
575			if (_is_t1_higher_prio_than_t2(thread, t)) {
576				sys_dlist_insert_before(pq, &t->base.qnode_dlist,
577							&thread->base.qnode_dlist);
578				return;
(gdb) l
579			}
580		}
581	
582		sys_dlist_append(pq, &thread->base.qnode_dlist);
583	}
584	
585	void _priq_dumb_remove(sys_dlist_t *pq, struct k_thread *thread)
586	{
587		__ASSERT_NO_MSG(!_is_idle(thread));
588	
(gdb) s
575			if (_is_t1_higher_prio_than_t2(thread, t)) {
(gdb) bt
#0  _priq_dumb_add (pq=pq@entry=0x200019ac <_kernel+40>, 
    thread=thread@entry=0x2000064c <adv_thread_data>)
    at /home/ubuntu/zephyr/kernel/sched.c:575
#1  0x0001ab06 in _add_thread_to_ready_q (thread=0x2000064c <adv_thread_data>)
    at /home/ubuntu/zephyr/kernel/sched.c:311
#2  0x0001b4f8 in z_clock_announce (ticks=<optimized out>)
    at /home/ubuntu/zephyr/kernel/timeout.c:194
#3  0x00004bc2 in _isr_wrapper ()
    at /home/ubuntu/zephyr/arch/arm/core/isr_wrapper.S:120
(gdb) s
_is_t1_higher_prio_than_t2 (t2=0x2000064c <adv_thread_data>, 
    t1=0x2000064c <adv_thread_data>) at /home/ubuntu/zephyr/kernel/sched.c:91
91		if (t1->base.prio < t2->base.prio) {
(gdb) p t2->base.prio
$1 = -9 '\367'
(gdb) p t1->base.prio
$2 = -9 '\367'
(gdb) s
_priq_dumb_add (pq=pq@entry=0x200019ac <_kernel+40>, 
    thread=thread@entry=0x2000064c <adv_thread_data>)
    at /home/ubuntu/zephyr/kernel/sched.c:574
574		SYS_DLIST_FOR_EACH_CONTAINER(pq, t, base.qnode_dlist) {
(gdb) p pq
$3 = (sys_dlist_t *) 0x200019ac <_kernel+40>
(gdb) p pq->next 
$4 = (struct _dnode *) 0x20000b5c <prio_recv_thread_data>
(gdb) p pq->next->next
$5 = (struct _dnode *) 0x20000550 <tx_thread_data>
(gdb) p pq->next->next->next
$6 = (struct _dnode *) 0x2000064c <adv_thread_data>
(gdb) p pq->next->next->next->next
$7 = (struct _dnode *) 0x2000064c <adv_thread_data>
(gdb) p pq->next->next->next->next->next
$8 = (struct _dnode *) 0x2000064c <adv_thread_data>
(gdb) p t
$9 = (struct k_thread *) 0x2000064c <adv_thread_data>
(gdb)

xiaoliang314, 2019-01-25:
My version is:
commit 2caecb54a37dca72fdd20aa45ac95e4eeeaf0adb (HEAD -> master, origin/master, origin/HEAD)
Author: Jan Van Winkel <jan.van_winkel@dxplore.eu>
Date:   Sat Jan 19 19:38:50 2019 +0100

    display: Remove stale ILI9340 header file

    Remove stale ILI9340 header file from global includes.

    Signed-off-by: Jan Van Winkel <jan.van_winkel@dxplore.eu>

xiaoliang314, 2019-01-30:
When I pause with gdb, it stops on SYS_DLIST_FOR_EACH_CONTAINER, and the next of the adv_thread list node points to itself, causing SYS_DLIST_FOR_EACH_CONTAINER to become a dead loop. This seems to happen when adv_thread calls k_sleep.

nashif, 2019-01-30:
@andyross fyi

xiaoliang314, 2019-02-01:
Thanks,  Expect it to be fixed.

pirast, 2019-02-03:
I might also be running into this problem (under heavy load of Mesh) but am not so familar with gdb unfortunately. I have also noticed next elements pointing to the referring element itself, but in my case of wait_q -> waitq. The symptoms are that my application freezes (i.e. it does not send / process advertisement messages) but it does not print a kernel panic or such in the serial console. Is it possible that I am hit by the same bug? @xiaoliang314 does this sound familar to you? 

I'm on ff88b7f255a1ff5249fd178b4cab858862fe3d7f.

_pend_current_thread (key=0, wait_q=wait_q@entry=0x20001d88 <entropy_nrf5_data+24>, timeout=-1) at C:/Users/ma/zephyrproject/zephyr/kernel/sched.c:409
409	{

l
404		_ready_thread(th);
405	}
406	#endif
407	
408	int _pend_current_thread(u32_t key, _wait_q_t *wait_q, s32_t timeout)
409	{
410	#if defined(CONFIG_TIMESLICING) && defined(CONFIG_SWAP_NONATOMIC)
411		pending_current = _current;
412	#endif
413		pend(_current, wait_q, timeout);
l
414		return _Swap(key);
415	}
416	
417	struct k_thread *_unpend_first_thread(_wait_q_t *wait_q)
418	{
419		struct k_thread *t = _unpend1_no_timeout(wait_q);
420	
421		if (t != NULL) {
422			(void)_abort_thread_timeout(t);
423		}
s
411		pending_current = _current;
bt
#0  _pend_current_thread (key=0, wait_q=wait_q@entry=0x20001d88 <entropy_nrf5_data+24>, timeout=-1) at C:/Users/ma/zephyrproject/zephyr/kernel/sched.c:411
#1  0x00023c42 in _impl_k_sem_take (sem=sem@entry=0x20001d88 <entropy_nrf5_data+24>, timeout=timeout@entry=-1) at C:/Users/ma/zephyrproject/zephyr/kernel/sem.c:151
#2  0x00022c8a in k_sem_take (timeout=-1, sem=0x20001d88 <entropy_nrf5_data+24>) at zephyr/include/generated/syscalls/kernel.h:103
#3  entropy_nrf5_get_entropy (device=<optimized out>, buf=0x2000656a <_interrupt_stack+1970> "", len=2) at C:/Users/ma/zephyrproject/zephyr/drivers/entropy/entropy_nrf5.c:240
#4  0x00004076 in _impl_entropy_get_entropy (length=4, buffer=0x20006568 <_interrupt_stack+1968> "\200\067", dev=<optimized out>) at ../../../../include/entropy.h:76
#5  entropy_get_entropy (length=4, buffer=0x20006568 <_interrupt_stack+1968> "\200\067", dev=<optimized out>) at zephyr/include/generated/syscalls/entropy.h:13
#6  sys_rand32_get () at C:/Users/ma/zephyrproject/zephyr/subsys/random/rand32_entropy_device.c:31
#7  0x000007ea in button_pressed_worker (work=<optimized out>) at ../src/main.c:567
#8  0x00024320 in _timer_expiration_handler (t=0x2000005c <heartbeat_timer>) at C:/Users/ma/zephyrproject/zephyr/kernel/timer.c:69
#9  0x0002425e in z_clock_announce (ticks=<optimized out>) at C:/Users/ma/zephyrproject/zephyr/kernel/timeout.c:190
#10 0x00005532 in _isr_wrapper () at C:/Users/ma/zephyrproject/zephyr/arch/arm/core\isr_wrapper.S:120
#11 <signal handler called>
#12 0x122003f6 in ?? ()
#13 0x1e325c5a in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
s
413		pend(_current, wait_q, timeout);
s
pend (thread=0x20000edc <recv_thread_data>, wait_q=0x20001d88 <entropy_nrf5_data+24>, timeout=-1) at C:/Users/ma/zephyrproject/zephyr/kernel/sched.c:339
339	{
p wait_q
$2 = (_wait_q_t *) 0x20001d88 <entropy_nrf5_data+24>
p wait_q->waitq
$3 = {{head = 0x20000edc <recv_thread_data>, next = 0x20000edc <recv_thread_data>}, {tail = 0x20000edc <recv_thread_data>, prev = 0x20000edc <recv_thread_data>}}
p wait_q->waitq->next
$4 = (struct _dnode *) 0x20000edc <recv_thread_data>
p wait_q->waitq->next
$5 = (struct _dnode *) 0x20000edc <recv_thread_data>
p wait_q->waitq->next->next
$6 = (struct _dnode *) 0x20000edc <recv_thread_data>
p wait_q->waitq->next->next->next
$8 = (struct _dnode *) 0x20000edc <recv_thread_data>

pirast, 2019-02-15:
Hi @xiaoliang314, thanks! I have seen some commits in master that may tackle the issue. However, I still seem to encounter it but am not able to reproduce it right now with a debugger attached (trying over the weekend). In case you have time: Could you cross-check? You seem to have a more appropriate test setup at hand.

xiaoliang314, 2019-02-18:
@pairst @andyross I have updated the code to the latest version. I seem to be harder to reproduce it, but it still has problems. I tested the pressure nearly 50,000 times but it still dies. I didn't use gdb to view the status, but the status is still that the uart console does not have any output.

xiaoliang314, 2019-02-18:
@pirast I modified the sample code so that it could be controlled by the host computer. I sent the next packet as soon as it was sent successfully. My uart baud rate is 1M.

pirast, 2019-02-21:
Ok, so I was able to reproduce it on latest master. This is the backtrace I could obtain with CONFIG_DEBUG=y. As before, no exception is printed on the serial console.

`#0  sys_dlist_peek_next (node=0x20000f10 <adv_thread_data+24>, list=0x200092d8 <timeout_list>) at /home/psmeshloc/zephyr/include/misc/dlist.h:341
No locals.
#1  next (t=0x20000f10 <adv_thread_data+24>) at /home/psmeshloc/zephyr/kernel/timeout.c:42
        n = <optimized out>
#2  0x0002c24c in _add_timeout (to=0x20000a40 <tx_thread_data+24>, fn=fn@entry=0x2b5e5 <z_thread_timeout>, ticks=ticks@entry=2) at /home/psmeshloc/zephyr/kernel/timeout.c:85
        t = <optimized out>
        __i = <optimized out>
        __key = {key = 32}
#3  0x0002bb60 in _add_thread_timeout (ticks=2, th=<optimized out>) at /home/psmeshloc/zephyr/kernel/include/timeout_q.h:46
No locals.
#4  _impl_k_sleep (duration=duration@entry=5) at /home/psmeshloc/zephyr/kernel/sched.c:897
        expected_wakeup_time = 1672158
        ticks = 2
        local_lock = <optimized out>
        key = <optimized out>
#5  0x00007350 in k_sleep (duration=5) at zephyr/include/generated/syscalls/kernel.h:21
No locals.
#6  on_failed_write (retry_cnt=retry_cnt@entry=3) at /home/psmeshloc/zephyr/subsys/logging/log_backend_rtt.c:148
No locals.
#7  0x000073ca in data_out_block_mode (data=0x20000770 <char_buf> "k buffers\033[0m\r\nr", length=15, ctx=<optimized out>) at /home/psmeshloc/zephyr/subsys/logging/log_backend_rtt.c:190
        ret = 0
        retry_cnt = 3
#8  0x00006b0e in buffer_write (outf=0x73a5 <data_out_block_mode>, buf=0x20000770 <char_buf> "k buffers\033[0m\r\nr", len=15, ctx=0x0 <crc32_ieee>) at /home/psmeshloc/zephyr/subsys/logging/log_output.c:134
        processed = 536874768
#9  0x000070b6 in log_output_flush (log_output=log_output@entry=0x2db58 <log_output>) at /home/psmeshloc/zephyr/subsys/logging/log_output.c:142
No locals.
#10 0x00007330 in panic (backend=<optimized out>) at /home/psmeshloc/zephyr/subsys/logging/log_backend_rtt.c:241
No locals.
#11 0x000066d0 in log_backend_panic (backend=<optimized out>) at /home/psmeshloc/zephyr/include/logging/log_backend.h:176
No locals.
#12 log_panic () at /home/psmeshloc/zephyr/subsys/logging/log_core.c:397
        i = 1
        backend = <optimized out>
#13 0x000091c8 in _Fault (esf=0x20004480 <tx_thread_stack+376>, exc_return=<optimized out>) at /home/psmeshloc/zephyr/arch/arm/core/fault.c:674
        reason = <optimized out>
        fault = 5
#14 0x00009336 in __usage_fault () at /home/psmeshloc/zephyr/arch/arm/core/fault_s.S:143
No locals.
#15 <signal handler called>
No symbol table info available.
#16 0xb8331624 in ?? ()
No symbol table info available.
#17 0xcaf31f5a in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
rrupt stack?)`

jhedberg, 2019-03-07:
I don't see how the scheduling corruption is Bluetooth related, so removing myself from the reviewers - at the same time I don't know who the right person would be so this will be unassigned for awhile.

laperie, 2019-03-07:
@andyross , assigning to you:  Please request reporter to re-verify once the fixes you work on are merged

andyross, 2019-03-07:
Is this still reproducible? Note that there have been several fixes to scheduler and locking code since the HEAD referenced above (dff6b7145 seems most likely) which could produce symptoms exactly like that.  Would be good to check again.  Or maybe it was a single spurious failure?  Has it happened since?

xiaoliang314, 2019-03-08:
@andyross I will update to the latest version for testing, I have prepared at least 12 hours of testing time and see if the problem exists.

bdrlamb, 2019-03-09:
I have run into this same infinite loop recently.  After doing some investigation, I believe in my case it is caused by k_spin_lock not masking all interrupts (Arch=STM32L1).  It appears both higher priority (31 is what I tested with) and IRQ_ZERO_LATENCY do not get masked. 

That can cause issues if the ISR interrupts _add_timeout or _remove_timeout, and the ISR itself calls k_poll_signal_raise or other kernel functions which call _add_timeout or _remove_timeout.  In my case, _add_timeout was being interrupted and other calls were being made to _add_timeout/_remove_timeout and it ultimately ended up breaking the linked list.  

I'm not sure if this is a bug or not but I think it might be worthwhile to have k_spin_lock assert if it catches itself being re-entrant.  Or at least have the kernel timer locally record whether or not the list is locked and having it bugcheck if it finds that it somehow has become re-entrant - this was a difficult to track down given how difficult it is to reproduce.

bdrlamb, 2019-03-09:
One other option may be to have a number of the kernel functions assert if they're ever called from an NMI.

xiaoliang314, 2019-03-11:
@bdrlamb Are you using the latest version?

andyross, 2019-03-11:
@bdrlamb: The rule is that _arch_irq_lock() (the underlying primitive used by the spinlocks) must mask any interrupt used to inspect or modify Zephyr kernel state.  It is allows that there be some levels higher than that, but they have to be for driver-specific purposes and only touch things that Zephyr never looks at.  Some of the radio code on nRF5x works this way that I know of.  The ZERO_LATENCY feature is deliberately placed at that priority.

And FWIW: when CONFIG_ASSERT is true, k_spin_lock() does indeed include a validation layer that will catch reentrant locking, and releases of unlocked locks.

andyross, 2019-03-12:
I'm going to close this one.  I'm really quite certain this is the issue resolved with commit dff6b714504c.  Please reopen if the symptom reappears.

closed 2019-03-12 by andyross

bdrlamb, 2019-03-28:
@xiaoliang314 I wasn't using the absolute latest version, but it definitely included commit dff6b71.  The issue was my fault (calling kernel functions in NMIs) and I have not encountered it since.