zephyrproject-rtos/zephyr #12728: docs: Hard to find guidelines for ext/ maintenance

pfalcon, 2019-01-25:
This comes from https://github.com/zephyrproject-rtos/zephyr/pull/12711#discussion_r250942649 .

I'd think a can find a link to give a new developer by grepping docs/* for `ext/`, but nope, that doesn't work. I had to do creative twisting of search terms to arrive at https://docs.zephyrproject.org/latest/contribute/ finally, which deals with that matter. Suggestion: make instructions there more concrete, specifying that in majority of cases, code imported form 3rd-party projects should be in `ext/` subtree.

nashif, 2019-01-26:
lets wait with this until dealing with external components is finalized now that we are moving to a multi-repo layout. But good point.

nashif, 2019-06-02:
This is now covered in the Modules section in the guide.

closed 2019-06-02 by nashif