zephyrproject-rtos/zephyr #12730: Add useful project description on Github

pfalcon, 2019-01-25:
Current Github description reads:

> Primary GIT Repository for the Zephyr Project

This is totally useless for folks who don't know what "Zephyr Project" is. Github project description should be seen as a basic means to spread the word about the project - the description is what people see in project search results for example.

Example of suggested replacement:

> Zephyr Project is a new generation, scalable, optimized, secure RTOS for multiple hardware architectures.

Which is an abbreviated version of first paragraph of README.

dbkinder, 2019-01-25:
Good idea, and your description works for me... but I don't have permission to alter that -- @nashif

nashif, 2019-01-26:
done

closed 2019-01-26 by nashif