zephyrproject-rtos/zephyr #12775: USB audio isochronous endpoints

m-nilsson, 2019-01-28:
Hi,

We would like support for USB audio and MIDI, mainly for the NRF52840.
The current implementation does not seem to support isochronous endpoints but it looks like they were included in the LPCUSB stack at some point, but not in a release branch.
https://sourceforge.net/p/lpcusb/code/HEAD/tree/trunk/

I have seen other people here mentioning it as well.
Would it be possible to include this functionality in Zephyr in the near future?
Is there any ongoing work on this?

pawelzadrozniak, 2019-03-04:
Hi.
I am currently working on audio class. I was able to modify testusb class by adding iso endpoints and pass the basic linux test on nrf52840.

m-nilsson, 2019-03-04:
@pawelzadrozniak 
Ok, that's great news!

raveslave, 2019-03-25:
Can we request this to support USB Audio Class 2.0 as well?

masz-nordic, 2019-03-28:
@raveslave Can you make a separate issue for Audio 2.0? I think @pawelzadrozniak is implementing 1.0.

benolayinka, 2019-05-17:
@pawelzadrozniak any luck?

raveslave, 2019-10-28:
@pawelzadrozniak just checking in if you planned on merging your work. would be awesome!

raveslave, 2019-11-21:
> Hi.
> I am currently working on audio class. I was able to modify testusb class by adding iso endpoints and pass the basic linux test on nrf52840.

do you have this in a repo somewhere?

emob-nordic, 2019-11-29:
> > Hi.
> > I am currently working on audio class. I was able to modify testusb class by adding iso endpoints and pass the basic linux test on nrf52840.
> 
> do you have this in a repo somewhere?

Hello, I've just get notification about this issue. Pawel is not working in the project anymore, I've recently start work in this topic. I don't know what Pawel had but [here](https://github.com/emob-nordic/zephyr/tree/usb_audio_class) is the temp branch where I push the changes. Feel free to comment.

emob-nordic, 2019-12-19:
#21526

PR #21526 (2020-05-08): usb: USB Audio class implementation
Tested on nRF52840 only, may not work for other platforms.

Two sample apps attached:
-> headphones + microphone
-> headset

Both apps perform loopback. See README.rst for details.

Fixes: #12775
~~Blocked by: #24735~~
closed 2020-05-08 by carlescufi