zephyrproject-rtos/zephyr #12800: topic-counter: nrfx_*: counter_set_top_value inconsistent behavior

pabigot, 2019-01-28:
The documentation does not indicate whether `counter_set_alarm()` which maps to `counter_set_top_value()` should be a one-shot or repeating capability.  The behavior for Nordic is inconsistent:

For RTC:

* If `counter_set_top_value` is configured with the maximum top value the callback is invoked once, but not rescheduled, though the callback remains associated with the channel.
* If `counter_set_top_value` is configured with a top value less than the maximum the RTC clock is cleared and the timer rescheduled, resulting in a repeated callback at the requested interval.

For TIMER:

* If `counter_set_top_value` is the callback is repeated at the requested interval without being rescheduled.

nashif, 2019-02-11:
@pabigot @nordic-krch  is this still an issue?

pabigot, 2019-02-11:
AFAIK it was closed by #12815.

PR #12815 (2019-01-29): drivers: counter: Fix default top value callback in counter_nrfx_rtc.c
In case of default top value, driver was calling top value
callback only on the first period. This was not expected behavior.

Fixes #12800.
closed 2019-02-11 by pabigot