zephyrproject-rtos/zephyr #12810: Up Squared serial console character output corrupted

fulong82, 2019-01-29:
**Describe the bug**
After rebased to latest master branch, observe serial console character output has corrupted.
Tried to revert commit below and found issue no longer there:
f0db0b8815 driver: uart_ns16550.c: necessary changes for arc iot soc

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=up_squared ../
3. make menuconfig
4. select CONFIG_SHELL_BACKEND_SERIAL, CONFIG_LOG
5. make
6. copy newly build zephyr.strip image to usb boot drive
7. boot up Up Squared with the boot drive
8. Observing console printing corrupted

**Expected behavior**
Console printing correctly.

**Impact**
Unable to interact with shell terminal.

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 14.04
 - Toolchain: Zephyr SDK
 - Commit: 5bd5e4e4406a56c61b4ad0609e3fe238e8ab6864

fulong82, 2019-01-29:
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/2806906/config.txt)

Forgot to attach the config file for reference.

Regards,
Wilson

nashif, 2019-01-30:
duplicate of #12652

closed 2019-01-30 by nashif