zephyrproject-rtos/zephyr #12814: Some Net Shell functions seems to not working when using NET_SOCKETS_OFFLOAD

AdamSzewczyk-Sigma, 2019-01-29:
**Describe the bug**
tcp connect seems to call net_context_offloading API functions even for aplications with NET_SOCKETS_OFFLOAD that leads to fall to simplelink_dummy_get handler.

**To Reproduce**
1.) build project for CC3220SF with CONFIG_NET_SHELL=y and CONFIG_NET_SOCKETS_OFFLOAD=y
2.) connect with serial terminal
3.) connect to wifi
3.) Invoke: net tcp connect your_server_adress your_server_port
4.) watch error messages

**Expected behavior**
Established tcp connection

**Screenshots or console output**
net tcp connect 192.128.0.101 4444
Connecting from 10.67.36.145:0 to 192.128.0.101:4444
Cannot get TCP context (-1)
uart:~$ [01:14:31.763,244] <err> wifi_simplelink: NET_SOCKET_OFFLOAD must be configured for this driver
[01:14:31.763,244] <err> wifi_simplelink: NET_SOCKET_OFFLOAD must be configured for this driver

**Environment (please complete the following information):**
 - OS: Linux
 - gnuarmemb
 - c2bea667641a6ccb75e1e7511ba8e8f42ef7eef9

jukkar, 2019-01-29:
The net-shell is still using net_context API to connect TCP. The net_context API does not support socket offloading so you see the error. The plan is to convert net-shell to use BSD socket interface eventually but we are not yet there. If you have spare time, patches are welcome to fix this.

pfalcon, 2019-02-04:
@vanti, @GAnthony : FYI

jukkar, 2020-01-28:
Going through old issues. This is a good candidate for closing. Can be re-opened if really needed.

closed 2020-01-28 by jukkar

mbrunnen, 2021-11-05:
Hi, I am having the same Problem. I wanted to use zperf on a CC3235 and ran into this.